platformio/platform-espressif32 #5: Oficial DOIT ESP32 support?

sabas1080, 2016-12-16:
Hi

I have test the board DOIT ESP32

http://www.banggood.com/ESP32-Development-Board-WiFiBluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board-p-1109512.html?rmmds=search

I upload example blink and I have this error

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3ffc0008,len:0
load:0x3ffc0008,len:1388
load:0x40078000,len:3608
load:0x40080000,len:260
entry 0x40080034
E (27) boot: nothing to load
user code done
```

Thanks

ivankravets, 2016-12-16:
Hi,

It's a known issue. Try to flash board with Arduino IDE (any sketch), then back to PIO. It should work. We will take a look later why it doesn't work.

sabas1080, 2016-12-17:
I try with Arduino IDE (1.6.11) and not work

sabas1080, 2016-12-17:
Change error
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3ffc0008,len:0
load:0x3ffc0008,len:1388
load:0x40078000,len:3608
load:0x40080000,len:260
entry 0x40080034
```

sabas1080, 2016-12-17:
Work upload with pio, example wifiscan

Thanks

closed 2016-12-17 by sabas1080

mintisan, 2017-01-06:
I also encounter the same issue with ESP32 Core Board v2

[image]

And Use Arduino Blink example to fix it : 

Note: when you download sketch, u should press boot button.

```
int ledPin = 5;

void setup()
{
    pinMode(ledPin, OUTPUT);
    Serial.begin(115200);
}

void loop()
{
    Serial.println("Hello, world!");
    digitalWrite(ledPin, HIGH);
    delay(500);
    digitalWrite(ledPin, LOW);
    delay(500);
}
```

[image]

then use platform to download sketch again(no need to press boot button), then everithing is fine now

```
/*
 * Blink
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */

#include <Arduino.h>

#define LED_BUILTIN 13

void setup()
{
        // initialize LED digital pin as an output.
        pinMode(LED_BUILTIN, OUTPUT);
        Serial.begin(115200);
        Serial.println("setup");
}

void loop()
{
        Serial.println("loop");
        // turn the LED on (HIGH is the voltage level)
        digitalWrite(LED_BUILTIN, HIGH);
        // wait for a second
        delay(1000);
        // turn the LED off by making the voltage LOW
        digitalWrite(LED_BUILTIN, LOW);
        // wait for a second
        delay(1000);
}

```

[image]

dagrende, 2017-01-21:
here is a detailed description on how to get DOIT ESP32 working according to the information above: [http://dagrende.blogspot.com/2017/01/how-to-use-doit-esp32-devkit.html](http://dagrende.blogspot.com/2017/01/how-to-use-doit-esp32-devkit.html).