platformio/platform-espressif32 #10: can't support c99 for now?

mintisan, 2017-01-13:
When I try [espidf-http-request](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-http-request), error come out below:

```
[Fri Jan 13 22:28:23 2017] Processing esp32dev (platform: espressif32, board: esp32dev, framework: espidf)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 compatible libraries
Looking for dependencies...
Project does not have dependencies
Compiling .pioenvs/esp32dev/src/main.o
src/main.cpp:50:30: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]

"Host: "WEB_SERVER "\n"
^
src/main.cpp: In function 'void initialise_wifi()':
src/main.cpp:88:9: error: C99 designator 'ssid' outside aggregate initializer
};
^
src/main.cpp:88:9: error: C99 designator 'password' outside aggregate initializer
src/main.cpp: In function 'void http_get_task(void*)':
src/main.cpp:100:9: sorry, unimplemented: non-trivial designated initializers not supported
};
^
src/main.cpp:100:9: sorry, unimplemented: non-trivial designated initializers not supported
Compiling .pioenvs/esp32dev/esp32/cpu_start.o
Compiling .pioenvs/esp32dev/esp32/deepsleep.o
*** [.pioenvs/esp32dev/src/main.o] Error 1
Compiling .pioenvs/esp32dev/esp32/event_default_handlers.o
========================== [ERROR] Took 2.51 seconds ==========================
```

seems not support code like:

```
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_WIFI_SSID,
            .password = EXAMPLE_WIFI_PASS,
        },
    };
```

ivankravets, 2017-01-13:
See

* http://docs.platformio.org/en/latest/projectconf.html#build-unflags
* http://docs.platformio.org/en/latest/projectconf.html#build-flags

```ini
[env:myenv]
build_unflags = -std=gnu++11
```

mintisan, 2017-01-16:
@ivankravets It doesn't work for me after I add `-std=gnu++11` to my platformio.ini

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
build_unflags = -std=gnu++11
```

And I doesn't see how can I specific standard in build_flags, like: 

```
build_flags = -std=c99
```

Do I skip something ?

Thank you for your given reference.

mintisan, 2017-01-17:
I use memset/memcpy to replace:

```
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_WIFI_SSID,
            .password = EXAMPLE_WIFI_PASS,
        },
    };
```

see https://github.com/platformio/platform-espressif32/pull/14

closed 2017-01-17 by mintisan