platformio/platform-espressif32 #8: undefined reference to `app_main'

mintisan, 2017-01-13:
OS : macOS 10.11.6
PlatformIO: 1.7.1(IDE), 3.2.1(Core)

use [espidf-hello-world](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-hello-world) example, then come error below(I can use Arduino example normally): 

```
Indexing .pioenvs/esp32dev/libtcpip_adapter.a
Linking .pioenvs/esp32dev/firmware.elf
.pioenvs/esp32dev/libesp32.a(cpu_start.o):(.literal.main_task+0x0): undefined reference to `app_main'
.pioenvs/esp32dev/libesp32.a(cpu_start.o): In function `main_task':
/Users/linjinhui/.platformio/packages/framework-espidf/components/esp32/cpu_start.c:169: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp32dev/firmware.elf] Error 1

========================== [ERROR] Took 15.54 seconds ==========================
```

mintisan, 2017-01-13:
should add `extern "C"` in front of `app_main`

```
/* Hello World Example
   This example code is in the Public Domain (or CC0 licensed, at your option.)
   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
 */
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"

void hello_task(void *pvParameter)
{
        printf("Hello world!\n");
        for (int i = 10; i >= 0; i--) {
                printf("Restarting in %d seconds...\n", i);
                vTaskDelay(1000 / portTICK_RATE_MS);
        }
        printf("Restarting now.\n");
        fflush(stdout);
        system_restart();
}

extern "C" void app_main()
{
        nvs_flash_init();
        system_init();
        xTaskCreate(&hello_task, "hello_task", 2048, NULL, 5, NULL);
}

```

closed 2017-01-13 by mintisan

Gryffoon, 2017-05-12:
I had the same issue even if I've added an extern "C" 
with no result.

The problem was that the component.mk  was at the wrong place. It must be under the main folder. And not near to 'Makefile' file.

Buckler89, 2017-12-29:
@Navis-Raven How do you use the comoponent.mk in platformio? For me seems that platformio just ignore it.

Navis-Raven, 2017-12-29:
I don't know, I was using Eclipse

derasd, 2018-09-27:
I had this error when I switched from compiling with platformio (vscode) to compiling with mingw32. The problem was that I had the code in the ./src folder instead of the ./main folder.