platformio/platform-espressif32 #11: partition table offset for flashing is apparently wrong

bleader, 2017-01-15:
From my understanding, the partition table on ESP32 is supposed to be 0x8000 based on [the documentation](http://esp-idf.readthedocs.io/en/latest/partition-tables.html):

> A single ESP32’s flash can contain multiple apps, as well as many different kinds of data (calibration data, filesystems, parameter storage, etc). For this reason a partition table is flashed to offset 0x8000 in the flash.

In [build/main.py:119](https://github.com/platformio/platform-espressif32/blob/develop/builder/main.py#L119) and the following cases for each framework, the flash address for partition table is 0x4000.

When flashing in verbose I can see the line calling esptool:

```
pio run -v -t upload
[...]
"/usr/bin/python2" "/home/bleader/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32 --port "/dev/ttyUSB0" --baud 115200 write_flash -z --flash_mode dio --flash_freq 80m 0x1000 "/home/bleader/.platformio/packages/framework-arduinoespressif32/tools/sdk/bin/bootloader.bin" 0x4000 "/home/bleader/.platformio/packages/framework-arduinoespressif32/tools/sdk/bin/partitions_singleapp.bin" 0x10000 .pioenvs/esp32dev/firmware.bin
[...]
```

This leads to something like:

```
user code not found
```

I say *something like* because my following attempt was to flash manually calling the same line, but using 0x8000 for partition table:

```
"/usr/bin/python2" "/home/bleader/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32 --port "/dev/ttyUSB0" --baud 115200 write_flash -z --flash_mode dio --flash_freq 80m 0x1000 "/home/bleader/.platformio/packages/framework-arduinoespressif32/tools/sdk/bin/bootloader.bin" 0x8000 "/home/bleader/.platformio/packages/framework-arduinoespressif32/tools/sdk/bin/partitions_singleapp.bin" 0x10000 .pioenvs/esp32dev/firmware.bin
```

This worked, but now, when flashing using **pio run -t upload** partition table do not change, and the one at 0x8000 isn't erased, so code is still executed fine.

So any module with a matching partition table data written at 0x8000 won't exhibit the issue anymore. Which makes this a pretty low priority issue I guess, and explains why it wasn't seen before, I just faced it as I used a brand new dev board directly with platformio command line without any other test prior to it.

ivankravets, 2017-12-13:
Could you reproduce this issue again?

bleader, 2017-12-13:
With the current version of platformio and updated platform-espressif32 it is apparently fixed, the address passed is now 0x8000.
After a quick search It was fixed in  https://github.com/platformio/platform-espressif32/commit/36c8eb4f85ba279f72d730162e7f0b086177cee5#diff-85c06424f599ebef79138c4c8b4456ccL134

closed 2017-12-13 by bleader

svofski, 2020-08-17:
So the offset is hardcoded in the platform and cannot be altered?

I'm trying to change flash mode to QIO, which requires a larger bootloader. This calls for partition table offset. But I cannot do that cleanly because the offset is hardcoded to 0x8000 in espidf.py. Or can I?

bleader, 2020-08-18:
It was at the time of this issue,  haven't played around with this recently so I wouldn't know if there are work around beside doing it manually like I did in the original issue.

svofski, 2020-08-18:
It seems to still be the case. The problem is not fixed, it was only made less obvious to most users.

valeros, 2020-08-18:
As a workaround you can always override the final `UPLOADCMD` in an [extra script](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html) to meet your needs.

svofski, 2020-09-10:
@valeros thanks! It's okay as a workaround.

But I find it misleading that this issue is closed. The uploader should take partition table offset from sdkconfig.

svofski, 2020-09-11:
To someone stumbling upon this while frustrated by the partition table offset issue, add this clumsy script the project dir and add a line to platformio.ini like so
```
[env:esp32]
...
extra_scripts = fix_partitions_offset.py
```

```
Import('env')

import json
from os.path import join, isfile

def get_sdk_configuration():
    config_path = join(env.subst("$BUILD_DIR"), "config", "sdkconfig.json")
    if not isfile(config_path):
        print('Warning: Could not find "sdkconfig.json" file\n')

    try:
        with open(config_path, "r") as fp:
            return json.load(fp)
    except:
        return {}

sdk_config = get_sdk_configuration()

offset = sdk_config.get("PARTITION_TABLE_OFFSET", 0x8000)
print(f'--- PARTITION_TABLE_OFFSET = 0x{offset:04x} -----------------')
print(env['FLASH_EXTRA_IMAGES'])
extra_images = env['FLASH_EXTRA_IMAGES']
for i in range(len(extra_images)):
    if extra_images[i][1].endswith('partitions.bin'):
        extra_images[i] = ('{:#04x}'.format(offset), extra_images[i][1])

print(env['FLASH_EXTRA_IMAGES'])
print('-----------------------------------------------------')
```

This however would be a trivial fix here, where all information is already loaded, you just need to replace the hardcoded address:
https://github.com/platformio/platform-espressif32/blob/19cb8eb6d4c8ee4a52a5e8cc3884e93e768ad97c/builder/frameworks/espidf.py#L1069

Thanks for your attention!

valeros, 2020-09-15:
Hi @svofski ! I've added a way to override the default offsets. Here is an example with a custom value for partition table
```
[env:esp32dev]
platform = espressif32
framework = espidf
board = esp32dev
board_upload.partitions_offset = 0x9000
```
If there is no custom value is `platformio.ini`, then we fall back to the value specified in the sdk configuration.
Would be great if you could retest with the [upstream version](https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream) of the platform.
Thanks!