platformio/platform-espressif32 #15: esp-idf framework out of date

mgiambalvo, 2017-01-22:
I don't really know how platformio works, but it seems as though the version of esp-idf this framework uses is determined by http://dl.platformio.org/packages/manifest.json. Any chance we could get an update to the recently released [2.0-rc1](https://github.com/espressif/esp-idf/releases/tag/v2.0-rc1). If I wanted to, is there a way for me to specify the version of esp-idf that platformio uses?

ivankravets, 2017-01-22:
@valeros, could you update IDF? Thanks!

rogelioRuiz, 2017-01-26:
I'm having this problem while trying to stablish a MQTT connection:

https://github.com/espressif/arduino-esp32/issues/85?cm_mc_uid=42109979851614853705915&cm_mc_sid_50200000=1485395577

When I use Arduino IDE there is no problem, the problem is only with platformio.
I guess is related with the esp-idf update.

Kuse, 2017-01-31:
Is anywhere a description how updates to newer version of esp-idf could be done on a local machine? The current version is very outdated and buggy.

ivankravets, 2017-01-31:
@valeros Could you describe in a few words how to update IDF?

@Kuse IDF package is located in `~/.platformio/packages/framework-espidf` folder.

ivankravets, 2017-02-21:
@mgiambalvo, @rogelioRuiz, @Kuse 

@valeros has deployed a new version of a builder for ESP IDF 2.0RC1. Could you test it?

```
pio platform uninstall espressif32
pio platform install https://github.com/platformio/platform-espressif32/archive/develop.zip
```

See new examples (BLE, etc.) - https://github.com/platformio/platform-espressif32/tree/develop/examples

ivankravets, 2017-02-22:
v0.6.0 is out!
https://github.com/platformio/platform-espressif32/releases/tag/v0.6.0

closed 2017-02-22 by ivankravets