platformio/platform-espressif32 #16: platform.io usering espidf framework , fatal error: sdkconfig.h not found

jmurphy644, 2017-02-03:
Issue: When attempting to build an esp32 project with espidf framework from platform.io, I receive the message " fatal error: sdkconfig.h not found" and compilation stops.   I am able to compile and flash successfully with the native ESP-IDF but I really would like to be able to use platform.io to do my work.  However, when trying to build the sample program, I get the error as shown below.   Perhaps it's just something I have mis-configured but since I am new to platform.io, I have no clue.   (Hardware is sparkfun esp32  thing)
================
platformio run
Compiling .pioenvs\esp32thing\bootloader\flash_encrypt.o
Compiling .pioenvs\esp32thing\bootloader\secure_boot.o
Compiling .pioenvs\esp32thing\bootloaderLog\log.o
Compiling .pioenvs\esp32thing\bt\bt.o
Compiling .pioenvs\esp32thing\driver\gpio.o
C:\users\joe\.platformio\packages\framework-espidf\components\esp32\ld\esp32.ld:17:23: fatal err
or: sdkconfig.h: No such file or directory

compilation terminated.
In file included from C:\users\joe\.platformio\packages\framework-espidf\components\freertos\inc
lude/freertos/FreeRTOS.h:98:0,
from src\main.c:1:
C:\users\joe\.platformio\packages\framework-espidf\components\freertos\include/freertos/FreeRTOS
Config.h:73:23: fatal error: sdkconfig.h: No such file or directory
compilation terminated.
In file included from C:\users\joe\.platformio\packages\framework-espidf\components\bootloader\s
rc\main\secure_boot.c:19:0:
C:\users\joe\.platformio\packa
(...continues in a similar manner....)

ccarrizosa, 2017-02-16:
Hi,
I had the same problem and I've fixed it copying sdkconfig.h file to src folder.

jmurphy644, 2017-02-16:
Hmmmm,  I think that I tried that but let me try again.

ivankravets, 2017-02-21:
Each project should have `sdkconfig.h`. See examples https://github.com/platformio/platform-espressif32/tree/develop/examples

closed 2017-02-21 by ivankravets