platformio/platform-espressif32 #19: ESP32 new IDF version release support...

madscientist42, 2017-03-27:
Espressif's released the RC2 of the 2.0 version of the SDK. 

Someone should update the platform stuff there to use the new stuff.  Seems to build fine with someone turning on all the key things in sdkconfig.h from their Linux menuconfig build framework and re-syncing the project after adding: 

join(FRAMEWORK_DIR, "components", "aws_iot", "include"),
join(FRAMEWORK_DIR, "components", "aws_iot", "port"),

...to the espidf.py file and re-syncing the projects you're building with PlatformIO for that target as they added a new set of headers.

ivankravets, 2017-03-27:
@valeros could you take a look? Thanks!

madscientist42, 2017-03-27:
Thanks guys, it's easy/simple and it should probably take about as long as it took for me to implement.

ivankravets, 2017-03-27:
@madscientist42 Valerii should check the whole changes in IDE RC2

madscientist42, 2017-03-27:
Yes.  Forgot to mention that one.  Issue's been edited to reflect the builder hook changes I missed when I made the original request.    There was one search path for includes missing.    

As an aside, it looks like that there's no checking within this framework for someone's settings for the IDF config- it's building everything which isn't as good as it should be.  I should be able to build an IDF project with all or less than all of the items turned on.  It gives more SRAM profile for some things if we turn off, say, WIFI if I'm doing BLE or BT in general things only.

projectgus, 2017-03-30:
Hi everyone,

ESP-IDF developer here. Just wanted to pop up and say we love that platform.io has integrated support for ESP-IDF, and we're keen to help out in any way we can to make it smooth for platform.io devs to continue to pull in support, updates, etc.

I have a couple of suggestions, although they're pretty poorly formed at this stage:

Looking at the [ESP-IDF Scons file](https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/espidf.py), it seems like something that could help would be automatically pulling component paths & layout from IDF into platform.io, rather than having to hard-code the list. I was originally going to suggest adding make targets to dump some of this info, but as platform.io users probably don't always have "make" I guess it would make more sense to parse some of this info in Python by examining the IDF source tree, component.mk files, etc. Most of that would be pretty straightforward, although there are some unusual components that are probably easier to hard-code in espidf.py. If there's anything we can do to tweak the layout of these files for easier parsing, let us know.

The situation for managing config seems more complex. Short of also parsing Kconfig files and generating sdkconfig.h, I can't think of any great suggestions. :/

Anyhow, I mostly just wanted to establish a line of contact. If there's anything we can tweak in ESP-IDF that would make platform.io integration easier, please let us know!

Angus

projectgus, 2017-03-30:
The other thing we've been talking about with other teams in similar situations (Arduino, Micropython) is to add a "make standalone" target, which takes an IDF source tree and a sdkconfig file and spits out a single compiled "lib" directory and a single "include" directory. This greatly simplifies integration into other build systems, although you lose a lot of flexibility.

ivankravets, 2017-03-30:
Hi @projectgus ,

Thanks a lot that contacted us. As you've spotted, the No.1 problem is `make`. I don't say that it is bad, I mean that it's not installed by default. That is why PIO Build System is based on SCons utility. We don't have any requirements to OS, only Python that is installed everywhere. If Windows user doesn't have Python, the new PIO IDE 2.0 uses own built isolated version of Python. PIO Build System is a super fast and works everywhere: Win/Unix.

> If there's anything we can do to tweak the layout of these files for easier parsing, let us know.

The answer is simple! We are ready to move [this script](https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/espidf.py)  to ESP-IDF repository where you will maintain it. We [did the same](https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py) with ESP32 for Arduino. Then, we will load your PIO Build script from the source code. See [PIO Build Script for Arduino&ESP32](https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/arduino.py).

I think it should be easy for you maintain it. Also, you can add extra tests to your CI config which will use PIO to build examples. If you will have any questions related to PIO Build System, you ping me or @valeros and we would be glad to help.

 > The situation for managing config seems more complex.

How about to create simple CLI wizard? We use [click](http://click.pocoo.org/5/) package and it's installed with PIO. You can use in your Py file `import click`. So, you can create multiple [Prompts](http://click.pocoo.org/5/prompts/) and ask developer all information, finally, write `sdkconfig.h`.

From PIO Core side it will look as:
```
pio run --target configure
```

In build script you define new target`configure` (I'll explain how) and call extra click-based wizard and gather all information. Finally, you know all information about Project and Build Environment, and can overwrite `sdkconfig.h` in `env.subst(join("$PROJECTSRC_DIR", "sdkconfig.h"))`.

Later, we can even create GUI for this Wizard in PIO IDE. 

Where is profit? No need to bother with `make libncurses-dev flex bison gperf` and dozens other complicated steps. If the end developer just needs a CLI, simple command:
```
pip install platformio
# cd to empty project dir
platformio init --board esp32dev --framework espidf --template basic.blink
pio run -t upload
```

Board blinks. With PIO IDE 2.0 it's more simple.

>  "make standalone" target, which takes an IDF source tree and a sdkconfig file and spits out a single compiled "lib" directory and a single "include" directory. 

That will not work well with PIO. We need all source code of a framework. See [our latest Twitter-post](https://twitter.com/PlatformIO_Org/status/847151190488104961). We work on Debug feature within PIO IDE 2.0. It will be the first IDE with a simple "Debug" button which doesn't require any configuration. No need to configure debug interfaces, adapters, start GDB servers, etc. You create a project, select a board - PIO does other complex work for you automatically.

If you have any questions, don't hesitate to write here.

projectgus, 2017-03-30:
Hi Ivan,

Great to hear from you.

> The answer is simple! We are ready to move this script to ESP-IDF repository where you will maintain it. We did the same with ESP32 for Arduino

Unfortunately we're not in a position to do this. I see that it makes sense for Arduino & Platform.IO given they share a common history and a close relationship. However we have a number of projects building on top of IDF (Arduino, Micropython, NodeMCU, Espruino, Mongoose, probably others I can't think of right now.) We can't maintain everyone's build script for them, at least not if we plan to get any of our own work done!

What we can do is find ways to produce metadata or other interim files in a format which is easier for third party build systems to integrate. Things like include paths, flags, source file lists, etc. There's no reason why updating to a new IDF version shouldn't be as simple as a PIO developer (or any other project's developer) running a make target which spits out some generic information in a well-formed file, and then copying that metadata across to be loaded by the other build system. This way updating PIO for a new IDF can be as simple as committing a new metadata file, rather than having to manually update the build script. What do you think, would this be helpful?

>>    "make standalone" target, which takes an IDF source tree and a sdkconfig file and spits out a single compiled "lib" directory and a single "include" directory.
> That will not work well with PIO. We need all source code of a framework.

Nice debugging example! Very cool.

This makes sense. For the record, there's no reason why you can't have precompiled library files with embedded debug information and a matching source tree, so it's still possible to see the framework source code when debugging. However if you want to be able to edit the framework source code while debugging and recompile it (and I imagine you do), then that's a showstopper. No worries at all, makes perfect sense.

Angus

ivankravets, 2017-04-01:
> Nice debugging project! Very cool.

Hope to see ESP32 support soon 😊 Could someone from your team help us with testing? We will add support for a few adapters which ESP32 docs recommend, the only one thing is to re-test it with PIO IDE 2.0. Thanks!

Support for ST STM32 is already announced! https://community.platformio.org/t/platformio-debugging/1704

> What we can do is find ways to produce metadata or other interim files in a format which is easier for third party build systems to integrate. 

I got you, not a problem. You saw our build script and understand it's logic. Now, any format could be useful for us. Please ping me when it will be ready, we will rewrite our build script. I propose to use simple format, for example, JSON or etc. Easy for parsing/reading/editing.

projectgus, 2017-04-03:
> Could someone from your team help us with testing? We will add support for a few adapters which ESP32 docs recommend, the only one thing is to re-test it with PIO IDE 2.0. Thanks

I've asked around to see if anyone who is more familiar with JTAG has some time to put it through its paces.

> Now, any format could be useful for us. Please ping me when it will be ready, we will rewrite our build script. I propose to use simple format, for example, JSON or etc. Easy for parsing/reading/editing.

OK, will do. It may be a little while before we have the opportunity to schedule this work. For 2.0, at least, I'd recommend continuing to update the build script in the way you are now. But I'll let you know as soon as we have something.

ivankravets, 2017-04-12:
v0.7 is out with a support for the ESP-IDF 2.0 release

* https://github.com/platformio/platform-espressif32/releases/tag/v0.7.0

closed 2017-04-12 by ivankravets

mutex36, 2018-03-04:
Is ESP-IDF 3.0 support planned?