platformio/platform-espressif32 #20: Enhancement: Make the IDF config be managed within the project somehow

madscientist42, 2017-03-27:
Right now, it really looks like the IDF is all-on or not.  It's building all sorts of things that probably ought to be turned off by some means in the project itself.  This way I can specify things like WIFI off, etc.

boneskull, 2018-02-11:
You should be able to disable it in `src/sdkconfig.h`, shouldn't you?

Goldenstarc, 2018-12-09:
> Sorry for dummy question: I can't understang, how I can change sdkconfig.h values?
> Sdkconfig.h generated when building.
> 
> * build_flags in platformio.ini doesn't work.
> * sdkconfig.defaults also unused =(.

 its just SDKCONFIG with  .h extensions
but it look likes Platformio doesn't support customized SDKCONFIG :(

nevercast, 2019-01-07:
Is there a temporary workaround for this? Did you say Platform.IO tramples `src/sdkconfig.h`?

Goldenstarc, 2019-01-07:
> Is there a temporary workaround for this? Did you say Platform.IO tramples `src/sdkconfig.h`?
> 
> Edit: It seems that the sdkconfig.h is at least present for the espidf framework, but absent entirely from the esp32-arduino framework, I would like sdkconfig.h support in the esp32-arduino framework too, is there another repository that should be raised in?

just use IDF !
its the best.

Goldenstarc, 2019-01-25:
> A temporary solution is to configure options directly in `src/sdkconfig.h`. See example https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-hello-world/src/sdkconfig.h
> 
> Please note that the Espressif Team works on moving to `cmake` build system and Kconfig will be dropped.

Whoops
And cmake solve most windows guys problems.
What about implementing manuconfig of make?!

versamodule, 2019-10-31:
Is there any update to this? I have been using Visual Studio Code With ESP-IDF installed manually. I use this command to open menuconfig. 
idf.py menuconfig

So i am not sure why this would be so hard to implement. 

[image]

Kot-Zakhar, 2019-11-10:
Currently facing similar problem as in the topic.
Recently found, that defined in sdkconfig.h macroses can't be changed anyhow besides manual rewriting.
i'm trying to change the priorities of BT and WIFI, while coexisting. So CONFIG_SW_COEXIST_* marcoses need to be overriden.
I'm really afraid of things can happen, when someone (not even knowing) will try to build my code using fresh framework with not properly defined macroses and spend 2 or more days trying to find solution of this bug (like i did).

Also i'm really curios about how can i force a framework to regenerate sdkconfig.h file (doesn't really metter to what state: default or following some rules).

So in case someone faced that issue or knows the solution: answer me plz :)
Thanks.

moranfarky, 2020-01-28:
> A temporary solution is to configure options directly in `src/sdkconfig.h`. See example https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-hello-world/src/sdkconfig.h [...]

Modify sdkconfig.h works for someone ???

versamodule, 2020-02-15:
Can we get an update about this please ? 
PlatformIO is awesome, I want to switch to it fully but the lack of getting to menuconfig is a serious hindrance to working with ESP-IDF.

ivankravets, 2020-02-15:
We work on ESP-IDF 4.0 which has Python-based tool for reading menu config files. I hope we will add a new target `pio run -t menuconfig`.

moranfarky, 2020-02-16:
When should we expect to ESP-IDF 4 support ( and the menuconfig feature ) in platformIO

me21, 2020-02-16:
They said later this month on forum

moranfarky, 2020-02-16:
Great news , Thanks.

valeros, 2020-03-18:
Starting with ESP-IDF v4.0 we use the native configuration system. Could you please try your project with the upstream version of the platform, for example:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = espidf
```
Also take a look at updated docs for ESP-IDF:
https://docs.platformio.org/en/latest/frameworks/espidf.html

Thanks!

closed 2020-03-21 by ivankravets

z3cko, 2021-05-08:
Why has this been closed? As far as I understand this still does not work with platformio 4.2.1

valeros, 2021-05-10:
@z3cko What's your problem exactly?

maheshrajrp, 2024-01-13:
To any in future try this,
Open Terminal using PlatformIO New Terminal Icon/Command from VSCode, then 
"pio run -t menuconfig"

this should bring up menuconfig, if you are within some different menu, press escape should bring it back to home.