platformio/platform-espressif32 #21: ESP Build configuration should default to enabling the stack canary.

fake-name, 2017-04-16:
It seems like the compilation flags for ESP32 projects differ from the defaults from the arduino-esp32 repository. In particular, arduino-esp32 apparently defaults with the stack-canary option turned on, while the platformio build has it off.

I discovered this when trying to debug an extremely confusing issue where I had a stack overflow that was corrupting the freertos internal task management structures.

Principally, it seems that projects need to be compiled with `-D CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY`.

See: https://github.com/espressif/arduino-esp32/issues/315

ivankravets, 2017-04-21:
@valeros could you check that?

ivankravets, 2017-04-24:
I've just updated `espressif32` development platform. Please `pio update`.

ivankravets, 2018-05-25:
Fixed in the latest upstream version http://docs.platformio.org/en/latest/platforms/espressif32.html#upstream

closed 2018-05-25 by ivankravets