platformio/platform-espressif32 #22: Add support for OTA

ivankravets, 2017-04-27:
Moved from https://github.com/platformio/platformio-core/issues/945#issuecomment-297799459

valeros, 2017-04-28:
Hi @GanjaFarmer, @pikolo84! OTA feature is now in the development branch, could you please check it out?

GanjaFarmer, 2017-04-28:
Thanks, but I tried, I updated everything, and no changes, still errors:
```Linking .pioenvs\esp32dev\firmware.elf
Checking program size
text       data     bss     dec     hex filename
398993    52304   27576  478873   74e99 .pioenvs\esp32dev\firmware.elf
Building .pioenvs\esp32dev\firmware.bin
esptool.py v2.0-beta1
Looking for upload port...

Use manually specified: 192.168.0.14
Uploading .pioenvs\esp32dev\firmware.bin
esptool.py v2.0-beta2
Traceback (most recent call last):
File "C:\Users\Guillaume\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\esptool.py", line 2294, in <module>
main()
File "C:\Users\Guillaume\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\esptool.py", line 2057, in main
esp = chip_class(args.port, initial_baud)
File "C:\Users\Guillaume\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\esptool.py", line 166, in __init__
self._port = serial.serial_for_url(port)
File "c:\users\guillaume\.platformio\penv\lib\site-packages\serial\__init__.py", line 88, in serial_for_url
instance.open()
File "c:\users\guillaume\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 62, in open
raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '192.168.0.14': WindowsError(2, 'Le fichier sp\xe9cifi\xe9 est introuvable.')
*** [upload] Error 1```

It's trange that esptool is used, instead of espota.

valeros, 2017-04-29:
Hi, @GanjaFarmer !
Please make sure you are using the development version of the platform.
Run the next command from PlatformIO IDE Terminal:
`pio platform install https://github.com/platformio/platform-espressif32/archive/develop.zip`

GanjaFarmer, 2017-04-29:
Thanks @valeros this is exactly what I needed ;)

ivankravets, 2017-04-29:
@GanjaFarmer does it work for you?

GanjaFarmer, 2017-04-29:
Yes, it works nicely ;)

closed 2017-04-29 by ivankravets