platformio/platform-espressif32 #23: Gr eiainErr oe0pnce Itrutwttmoto P0) Error. and broken character

wjm0621, 2017-06-03:
I'm Using Sparkfun ESP32 Thing. and sending websocket message.
At the shown image, I met Gr eiainErr oe0pnce Itrutwttmoto P0) Error....

It is not always raised. just somtime shown.
I can't find this at google.

What is this Error means??

And some character is broken. (broken character is English character)
At shown image, code line 146. It will print "TCP handshaking with Server Connected. "
But below the Code, at word "handshaking", 'k', 'n' are broke.
How can i fix this problem??

[image]
[image]

ImanolGo, 2017-06-18:
Were you able to solve it? I have exactly the same issue!

valeros, 2017-06-19:
Hi @wjm0621 @ImanolGo !
Which version of framework do you use? Stable or staging? Do you observe the same behavior in Arduino IDE?

wjm0621, 2017-06-20:
I'm using Atom 1.17.2 version. and espressif32 package 0.7.4 version(latest).

ImanolGo, 2017-06-20:
I'm using the Arduino core for the ESP32

https://github.com/espressif/arduino-esp32

closed 2018-02-08 by ivankravets