platformio/platform-espressif32 #24: Arduino.h not found - ESP32 with ESP-IDF Framework

anilanilb, 2017-06-10:
Hi,
Copying issue from https://community.platformio.org/t/arduino-h-not-found-esp32-with-esp-idf-framework/1257 to platform-espressif32.

Currently I am using an ESP32 with the IDF Framework.
According to the docs https://github.com/espressif/arduino-esp3217 (section: Using as ESP-IDF component) the IDF Framework (supported by Platform IO and set up in the platformio.ino file) has an Arduino Core implemented and can be called by the ´initArduino()´ function. But compiler throws an error that Arduino.h cannot be found.

Has this been solved? if not, hope somebody can help here.

------------- Log Start -----------------
Processing esp32dev (platform: espressif32; lib_ldf_mode: deep; board: esp32dev; framework: espidf)
 
Verbose mode can be enabled via `-v, --verbose` option
Collected 0 compatible libraries
Looking for dependencies...
Project does not have dependencies
Compiling .pioenvs\esp32dev\src\main.o

Generating partitions .pioenvs\esp32dev\partitions_table.bin
Generating LD script .pioenvs\esp32dev\esp32_out.ld
Compiling .pioenvs\esp32dev\bootloader\bootloader_start.o
Compiling .pioenvs\esp32dev\bootloaderSupport\src\bootloader_flash.o
Compiling .pioenvs\esp32dev\bootloaderSupport\src\bootloader_random.o
Compiling .pioenvs\esp32dev\bootloaderSupport\src\efuse.o
Compiling .pioenvs\esp32dev\bootloaderSupport\src\esp_image_format.o
Compiling .pioenvs\esp32dev\bootloaderSupport\src\flash_encrypt.o
Compiling .pioenvs\esp32dev\bootloaderSupport\src\flash_partitions.o
src\main.c:8:52: fatal error: components/arduino/cores/esp32/Arduino.h: No such file or directory
compilation terminated.
*** [.pioenvs\esp32dev\src\main.o] Error 1
------------- Log End ------------------

Regards,
Andy

williamesp2015, 2017-07-04:
CHECK THIS OUT FOR PLATFORMIO
http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version

ivankravets, 2017-07-24:
https://github.com/espressif/arduino-esp3217 - broken link

Have you resolved this issue?

bryanmtdt, 2017-09-10:
Having the same problem here, I want to add components to the ESP-IDF but it seems that the arduino core is not been loaded.

Redferne, 2017-12-12:
Anyone succeeded with building Arduino as IDF component using PlatformIO?

ivankravets, 2017-12-13:
Could someone provide a source code to reproduce this issue?

ivankravets, 2017-12-29:
@me-no-dev is it our issue? Does it work with Arduino IDE?

BuzzRage, 2018-02-05:
same here, with WiFi.h, FS.h WString.h etc... 
It seems like something isn't well set. Maybe a path ? Maybe we have to precise `lib_extra_dirs` in the `platformio.ini` file ?

Everything works well in command line ([setting paths manually](https://esp-idf.readthedocs.io/en/latest/get-started/index.html), [cloning arduino repo as component ](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md?utf8=%E2%9C%93) etc..), but with PlatformIO in Atom, it failed. I tried to move required libraries everywhere without success.

Any news ?

me21, 2018-07-03:
I think `builder/frameworks/espidf.py` script is not setting include and source paths for Arduino framework used as ESP-IDF component. In fact, it isn't searching for `components` subfolder in the project directory at all.

I believe that Arduino as ESP-IDF component would be so useful. As far as I understand, many of the components located in `tools/sdk/lib` subfolder of Arduino framework are located in `components` subfolder of ESP-IDF and are built from source, making modifying them a breeze.

sticilface, 2019-07-02:
I am trying desperately to get this to work but the issues seem to be circular...  

Are there any clear examples of how to use arduino as a component of the IDF.

I currently get `fatal error: components/arduino/cores/esp32/Arduino.h: No such file or directory`

If I start including the relevant directories from Arduino-esp32 It just keeps spiralling with more and more requirements.. 

What I am really after is a way of enabling the core-dump feature of the IDF so that I can debug a crash... I'm heavily dependent on the Arduino framework though so cannot port it all to the IDF.  

Ideally I'd like the menuconfig options to be available in the arduino framework..  failing that as I understand it is WIP requiring the IDF to change to cmake... is there a way to get arduino to compile whilst using the idf?

I'm also a bit stuck as to how to configure menuconfig (where is the sdkconfig meant to me located for an idf project?)

atanisoft, 2019-07-02:
@sticilface please see PR #193. You should only have the sources for *YOUR* project in your project tree. All IDF bits and arduino-esp32 bits should come from the PIO frameworks.

MatheusAbel, 2019-07-03:
I'm having the same issue. I have an entire project which relies on 2/3 consolidated Arduino libraries and they doesn't have equivalent for espidf, so having Arduino as component working would save plenty of time of recoding those libs. 

Also my project needs ULP, so on PlatformIO only espidf is available. I have tried to clone @atanisoft commit from 07-02-2019 but couldnt manage to make it work too. I'd really be thankful with a solution to this.

atanisoft, 2019-07-03:
> Also my project needs ULP, so on PlatformIO only espidf is available. I have tried to clone @atanisoft commit from 07-02-2019 but couldnt manage to make it work too. I'd really be thankful with a solution to this.

What issue did you run into? I don't have any ULP+Arduino+IDF examples to work with to test this combination.

MatheusAbel, 2019-07-03:
> What issue did you run into? I don't have any ULP+Arduino+IDF examples to work with to test this combination.

My scenario is: I have a project which was written based on arduino framework. Now i want to implement ULP code to read sensors which is already done and working in a 'test project' lets say. 

As Platformio only supports espidf and I need to get my Arduino project working on idf the best way i've found is to use Arduino as a component o espidf as mentioned in this issue.  I get the same error than @anilanilb, Arduino cant be found on platformIO build.

I've followed the steps from [https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md](url) to add Arduino component. 

Also my sdkconfig.h contains 

> #define CONFIG_AUTOSTART_ARDUINO 1
> #define CONFIG_ENABLE_ARDUINO_DEPENDS 1

Any clue how could i make it work?

atanisoft, 2019-07-03:
> As Platformio only supports espidf and I need to get my Arduino project working on idf the best way i've found is to use Arduino as a component o espidf as mentioned in this issue. I get the same error than @anilanilb, Arduino cant be found on platformIO build.

Start with an IDF only app in PIO and use the sdkconfig.h from arduino-esp32 only. Apply my PR #193 to your local copy and build as a normal IDF project in PIO.

If you see failures in resolving arduino specific libraries try ```lib_compat_mode=strict``` in your platformio.ini file.

MatheusAbel, 2019-07-03:
> Start with an IDF only app in PIO and use the sdkconfig.h from arduino-esp32 only. 

What do you mean by sdkconfig.h from arduino-esp32 only? Would it be starting a project in PIO with arduino-esp32 platform and use that sdkconfig.h? Sorry if i didn't get it.

atanisoft, 2019-07-03:
Start with an IDF only project in PIO. Use the sdkconfig.h from arduino-esp32 since it includes the necessary definitions for arduino-esp32 (dependencies, etc). You don't need to add arduino-esp32 as a component when building with PIO and PR #193, the builder will take care of including the necessary components automatically with your project only containing *YOUR* files and not IDF/Arduino pieces.

MatheusAbel, 2019-07-03:
I managed to get it working. Thanks to @atanisoft which guided me. One thing I was missing is that I dont have to add the Arduino component, its just change de skdconfig and PIO makes all the hard work. 

Now starting to test if libs can work properly. I Will let you know.

valeros, 2019-07-04:
Hi @anilanilb @williamesp2015 @bryanmtdt @radaiko @Redferne @BuzzRage @sticilface @MatheusAbel! Finally, we've found some spare time and added initial support for this feature. Many thanks to @atanisoft who did a great job and provided a classy PR that served as the basis for the current implementation. Two new examples [blink](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-blink) and [wifiscan](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-wifiscan) from @atanisoft PR have been added to this repository. Now, the most of Arduino libraries should be compatible with IDF framework but keep in mind that this feature is still in an early experimental stage and IDF projects that need it must have a special version of `sdkconfig.h` file (see examples).

atanisoft, 2019-07-04:
@valeros I see a couple possible issues with the updated code:
1. arduino-esp32 has a few extra CCFLAGS which are not included in espidf.py when arduino is enabled, this may need to be reviewed further.
2. CPPDEFINES should also include F_CPU, ARDUINO_VARIANT and ARDUINO_BOARD
3. PIO_FRAMEWORK_ESP_IDF_ENABLE_EXCEPTIONS should likely be derived from CONFIG_CXX_EXCEPTIONS in sdk_params.

MatheusAbel, 2019-07-04:
@valeros As I said here:
> My scenario is: I have a project which was written based on arduino framework. Now i want to implement ULP code to read sensors which is already done and working in a 'test project' lets say.

It would be nice to have ULP support also on ArduinoFramework. I'm having a lot of trouble in order to use some Arduino libs on espidf since i2c driver from Arduino framework seems not to work well with espidf. 

To clarify, i'm using an i2c sensor which library was written for Arduino and so on it uses Wire and TwoWire libs to make i2c work. Now i'm struggling to make the things happen. I'll tell here when i have an answer

atanisoft, 2019-07-04:
> I'm having a lot of trouble in order to use some Arduino libs on espidf since i2c driver from Arduino framework seems not to work well with espidf.

Can you come on https://gitter.im/espressif/arduino-esp32 and we can discuss the issues you are having with the arduino-esp32 I2C system? It should have no issues with the IDF i2c drivers (provided you are not trying to use both simultaneously)

valeros, 2019-07-05:
@atanisoft Good points!
1. Just compared our build flags and flags from `make` log when Arduino is built as an IDF component and didn't notice any significant difference (for exceptions see 3). Am I missing something?
2. Same as (1), looks like the original build scripts don't add that macro, however `F_CPU` is already specified in our build.
3. `PIO_FRAMEWORK_ESP_IDF_ENABLE_EXCEPTIONS` config was added to allow users to enable exceptions even if they're using default `sdkconfig.h`. Do you think exceptions should be enabled by default for Arduino with ESP-IDF?

atanisoft, 2019-07-05:
@valeros

1. These flags are in the PIO build script in arduino-esp32 and also in platform.txt used by Arduino ide.
2. Same as (1).
3. I don't think they should be enabled always but instead allow the user to define it via sdkconfig (or menuconfig if they opt for that). In my PR I added a check for the sdkconfig entry in addition to the PIO_FRAMEWORK_ESP_IDF_ENABLE_EXCEPTIONS flag.

sticilface, 2019-07-05:
I've managed to get it to work.  

I needed to switch to the --dev branch of PIO
you have to use 
`platform = https://github.com/platformio/platform-espressif32.git`  for the upstream,

and crucially the arduino `#defines` are missing 
so I needed to add these

```cpp
  -D ESP32 
  -D ESP_PLATFORM 
  -D F_CPU=240000000L 
  -D ARDUINO=10805 
  -D ARDUINO_ARCH_ESP32 
  -D ARDUINO_VARIANT=\"d32_pro\" 
  "-D ARDUINO_BOARD=\"WEMOS LOLIN D32 PRO\"" 
  -D ARDUINO_ESP32_VER=\"2.10002.190628\" 
```

bizarrely not including `"` around the `ARDUINO_BOARD` resulted in this error

```
<command-line>:0:15: warning: missing terminating " character
```

which is a bit odd. 

Since upgrading to --dev -/+ using the above upstream esp32 I can no longer get @atanisoft environment to work.  

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_d32_pro.html
PLATFORM: Espressif 32 1.9.0 #a33c9fa > WEMOS LOLIN D32 PRO
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-espidf 3.30202.190627 (3.2.2), tool-esptoolpy 1.20600.0 (2.6.0)
Including Arduino-ESP32 from /Users/amelvin/.platformio/packages/framework-arduinoespressif32
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
```

It looks like might be using the wrong framework instead of his branch.  It fails trying to find the libs.

atanisoft, 2019-07-05:
@sticilface set framework to "espidf, arduino". Also for your defines, did you try just " and not \" ?

@valeros these missing defines are used by some libs and would be good to include for the Arduino IDF build.

atanisoft, 2019-07-05:
@valeros difference in PIO based on framework line:

espidf, arduino:
```
xtensa-esp32-elf-g++ -o .pio\build\esp32\src\ConfigurationManager.cpp.o -c -std=c++11 -fno-rtti -fno-exceptions -std=gnu++11 -Werror=all -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-error=unused-function -g3 -Os -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -DWITH_POSIX -DIDF_VER=\"3.30202.190627\" -DPLATFORMIO=40000 -DARDUINO_ESP32_DEV -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_INFO -DESP32_WIFI_MGR_MDNS_SOCKETPARAMS_LOG_LEVEL=VERBOSE -DESP32_WIFI_MGR_MDNS_QUERY_RES_LOG_LEVEL=VERBOSE -DARDUINOJSON_ENABLE_STD_STRING=1 -DARDUINOJSON_ENABLE_STD_STREAM=1 -D_GLIBCXX_USE_C99 -DLOCKED_LOGGING -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DGCC_NOT_5_2_0=0 @"C:\trains\ESP32CommandStation\.pio\build\esp32\longcmd-ab59af8556baaa0e11ed4fba0e94ddfd" src\ConfigurationManager.cpp
```

vs (arduino):
```
xtensa-esp32-elf-g++ -o .pio\build\esp32\src\ConfigurationManager.cpp.o -c -std=c++11 -fno-rtti -fno-exceptions -std=gnu++11 -g3 -Os -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -fexceptions
-DPLATFORMIO=40000 -DARDUINO_ESP32_DEV -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_INFO -DESP32_WIFI_MGR_MDNS_SOCKETPARAMS_LOG_LEVEL=VERBOSE -DESP32_WIFI_MGR_MDNS_QUERY_RES_LOG_LEVEL=VERBOSE -DARDUINOJSON_ENABLE_STD_STRING=1
-DARDUINOJSON_ENABLE_STD_STREAM=1 -D_GLIBCXX_USE_C99 -DLOCKED_LOGGING -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DARDUINO=10805 -DARDUINO_ARCH_ESP32 -DARDUINO_VARIANT=\"esp32\" "-DARDUINO_BOARD=\"Espressif ESP32 Dev Module\"" @"C:\trains\ESP32CommandStation\.pio\build\esp32\longcmd-8c6e2af4ed6f37384be531a108754512" src\ConfigurationManager.cpp
```

The entries: ARDUINO, ARDUINO_ARCH_ESP32, ARDUINO_VARIANT and ARDUINO_BOARD are missing. In the Arduino as IDF component make approach (not cmake) we can see a couple of these defined [here](https://github.com/espressif/arduino-esp32/blob/master/Makefile.projbuild#L7), the cmake version is missing these (issue to be filed). A simple fix for this is below:
```
diff --git a/builder/frameworks/espidf.py b/builder/frameworks/espidf.py
index 6f3cf45..3a97378 100644
--- a/builder/frameworks/espidf.py
+++ b/builder/frameworks/espidf.py
@@ -391,11 +391,16 @@ def build_arduino_framework():
         ]
     )

-    envsafe = env.Clone()
-    envsafe.Append(
-        CPPDEFINES=[("ARDUINO", 10805), ("ARDUINO_ARCH_ESP32", 1)]
+    env.Append(
+        CPPDEFINES=[
+            ("ARDUINO", 10805),
+            "ARDUINO_ARCH_ESP32",
+            ("ARDUINO_VARIANT", '\\"%s\\"' % env.BoardConfig().get("build.variant").replace('"', "")),
+            ("ARDUINO_BOARD", '\\"%s\\"' % env.BoardConfig().get("name").replace('"', ""))
+        ]
     )

+    envsafe = env.Clone()
     arduino_libs = []
     if "build.variant" in env.BoardConfig():
         envsafe.Append(

```

By having the CPPDEFINES only in the envsafe clone that would allow these only to be used by the arduino code and not user code (or user libraries under lib)

atanisoft, 2019-07-05:
@valeros found one other issue, LDF will need an update as well:
```
Processing esp32 (framework: espidf, arduino; platform: https://github.com/platformio/platform-espressif32.git; board: esp32dev)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 #a186f66 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), framework-espidf 3.30202.190627 (3.2.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Looking for ESP32 Async UDP library in registry
Warning! Library `{'requirements': None, 'name': 'ESP32 Async UDP'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'ESP32 Async UDP'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 35 compatible libraries
```
"ESP32 Async UDP" is a built in library from arduino-esp32 ([here](https://github.com/espressif/arduino-esp32/tree/master/libraries/AsyncUDP)) and the library.json added by PIO only lists "arduino" as a framework so it is excluded due to not an exact match.

When LDF runs, can it split the framework and test each "framework" to see if the library is "valid" ?

valeros, 2019-07-05:
Thanks, @atanisoft! I've updated the build script, now it should look better. As for the IDF, I tried a simple example from `ESP32 Async UDP` library and everything works as expected. I see there is a warning in your log, probably because you explicitly specified the dependency on the library. Could you please put together a simple project to reproduce this behavior, because LDF should automatically detect your dependency with the next configuration `framework = arduino, espidf`

atanisoft, 2019-07-05:
@valeros Yes, it was listed in "lib_deps" and also a dependency of ESPAsyncDNSServer (I've got a copy of this under lib with updates for esp32 support).

I'll pull the latest updates and give it another test.

oliverjc, 2019-07-13:
Please forgive my ignorance, I found myself here after needing to port my project over to esp-idf, how do I go about creating an esp-idf project within Platform IO (with Arduino as a component) with the option to use 'make menuconfig'.

There seem to be references to 'make menuconfig' above but how you go about using that with Platform IO is beyond my understanding.

sajeel-ahmed, 2019-07-21:
> I managed to get it working. Thanks to @atanisoft which guided me. One thing I was missing is that I dont have to add the Arduino component, its just change de skdconfig and PIO makes all the hard work.
> 
> Now starting to test if libs can work properly. I Will let you know.

hey @MatheusAbel how did you do this? Here is the problem i have been facing:
I have made a project on Arduino IDE by using arduino-esp32 libraries. Now, I have switched on PlatformIO and I want to make same project in PlatformIO by using ESP-IDF Framework. Actually, I was adding all Arduino libraries in the "lib" folder. but this is a hectic process.

Is it possible I can use Arduino esp32 core in ESP-IDF framework so I dont need to write whole libraries again.

I am new to PlatformIO so it is getting difficult for me to understand what to do. Your help will be really a solution for my problem.

sajeel-ahmed, 2019-07-21:
> Start with an IDF only app in PIO and use the sdkconfig.h from arduino-esp32 only. Apply my PR #193 to your local copy and build as a normal IDF project in PIO.
>
> If you see failures in resolving [...]

Hi @MikeDunston, I am in need of your help. I have made project in Arduino IDE using Arduino core of esp32. Actually i am working on esp32. Now i am shifting my code to PlatformIO and I have to use ESP-IDF framework. Actually, in Arduino IDE i have all the libraries which I have used in my code. 

I am copying all libraries in PlatformIO and it is not worth doing it. Is there any way I can use Arduino esp32 core in ESP-IDF Framework?? means so i can use libraries easily while working with esp-idf framework in PlatformIO. 

Thank you for your time and consideration. You help will definitely solve my problem. I was reading your previous discussion on this topic but I could not understand as i am learning PlatformIO.

Best regards
Sajeel Ahmed

tharunjoy, 2019-07-25:
Hi, I was able to setup and work with Arduino as ESP-IDF component while setting up the configuration as
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino, espidf

but I was facing some issues while I set the platform as espressif32 which is a stable build.
This is the error which I am facing.

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino, espidf)
-------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), framework-espidf 3.30202.190627 (3.2.2), tool-esptoolpy 1.20600.0 (2.6.0)

*** Multiple ways to build the same target were specified for: ........................./PlatformIO/platform-espressif32/examples/espidf-arduino-blink/.pio/build/esp32dev/partitions.bin  (from ['\.....platformio/packages/framework-arduinoespressif32/tools/partitions/default.csv'] and from ['\............/.platformio/packages/framework-espidf/components/partition_table/partitions_singleapp.csv'])
File ".................../.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 592, in <module>
============================================ [ERROR] Took 0.42 seconds ============================================
The terminal process terminated with exit code: 1

Can anyone please help me?

thewizkid87, 2019-07-31:
Hi all

@anilanilb  Thanks for the help, i have also got it to recognize the arduino mixed with esp-idf.

But i am also getting the same error as @tharunjoy 

I have downloaded the develop from here and am using the "espidf-arduino-blink" from the examples folder, with no changes or anything.

*** Multiple ways to build the same target were specified for: ...\Projects\espidf-arduino-blink\.pio\build\nina_w10\partitions.bin  (from ['...\\packages\\framework-arduinoespressif32\\tools\\partitions\\minimal.csv'] and from ['...\\Projects\\espidf-arduino-blink\\minimal.csv'])
File "...\platforms\espressif32\builder\frameworks\espidf.py", line 592, in <module>

I cleaned up the full path removing my personal data.

valeros, 2019-07-31:
Hi @thewizkid87 @tharunjoy ! Are you sure you are using the develop version of the platform?

thewizkid87, 2019-07-31:
@valeros 

I tried again, removed the platform then put it back.

Finally it works with these configs:

platform = https://github.com/platformio/platform-espressif32.git
framework = arduino, espidf
monitor_speed = 115200
board_build.partitions = partitions_singleapp.csv

Thanks for the help

ChristosPalisidis, 2019-08-07:
> @valeros As I said here:
>
>
> It would be nice to have ULP support also on ArduinoFramework. I'm having a lot of trouble in order to use some Arduino libs on espidf since i2c driver from Arduino [...]

@MatheusAbel i have the same problem. Do you have an answear?

txf-, 2019-08-08:
I tried compiling Some of the arduino examples, namely one that used httpclient. The sdkconfig.h file included in the examples from this repo. However that didn't work, was having tls issues and later fd errors from the client. 

I then tried using the sdkconfig from the official esp32 arduino repo. That didn't compile either, some aws configs were missing. So if anybody else is having similar issues just go grab the sdkconfig from [here](https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/include/config).

Then add these lines to it:

```
#define CONFIG_AWS_IOT_MQTT_TX_BUF_LEN 512
#define CONFIG_AWS_IOT_MQTT_RX_BUF_LEN 512
#define CONFIG_AWS_IOT_MQTT_NUM_SUBSCRIBE_HANDLERS 5
#define CONFIG_AWS_IOT_MQTT_MAX_RECONNECT_WAIT_INTERVAL 128000
#define CONFIG_AWS_IOT_MQTT_MIN_RECONNECT_WAIT_INTERVAL 1000
#define CONFIG_AWS_IOT_SHADOW_MAX_JSON_TOKEN_EXPECTED 120
#define CONFIG_AWS_IOT_SHADOW_MAX_SIZE_OF_THING_NAME 20
#define CONFIG_AWS_IOT_MQTT_HOST ""
#define CONFIG_AWS_IOT_MQTT_PORT 8883
#define CONFIG_AWS_IOT_SHADOW_MAX_SHADOW_TOPIC_LENGTH_WITHOUT_THINGNAME 60
#define CONFIG_AWS_IOT_SHADOW_MAX_SIZE_OF_THING_NAME 20
#define CONFIG_AWS_IOT_SHADOW_MAX_SHADOW_TOPIC_LENGTH_WITHOUT_THINGNAME 60
#define CONFIG_AWS_IOT_SHADOW_MAX_SIMULTANEOUS_ACKS 10
#define CONFIG_AWS_IOT_SHADOW_MAX_SIMULTANEOUS_THINGNAMES 10
#define CONFIG_AWS_IOT_SHADOW_MAX_JSON_TOKEN_EXPECTED 120
```

DUKELOLO, 2019-11-25:
Hello, 
I can compile examples and run fine but it only run arduino code, and never run the espidf part.
If i set #define CONFIG_AUTOSTART_ARDUINO 0 then it not compile (can not find setup and main arduino loop...)
How is possible to jump from arduino code to espidf part? or autostart with espidf? my code is espidf but i neet to use TFT_eSPI.h that is an arduino based library.
Thanks !!!

txgiao, 2021-03-19:
I successfully compile example "espidf-arduino-wifiscan" with setting CONFIG_AUTOSTART_ARDUINO=0 in sdkconfig.h. The problem is that during compiling, the  CONFIG_AUTOSTART_ARDUINO returns to 1, so actually I can not use Tasks. 
How can i set   CONFIG_AUTOSTART_ARDUINO=0 ?
Thanks

bwjohns4, 2022-04-26:
MIssing Arduino.h? What was the final resolution on this? I get this when trying to compile:

Scanning dependencies...
Dependency Graph
|-- <ESP32Ping> 1.5
|-- <NTPClient> 3.1.0
|-- <WifiManager> 0.15.0
Building in release mode
Compiling .pio/build/esp32devIDF/src/main.o
src/main.cpp:169:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://registry.platformio.org/search?q=header:Arduino.h
*
*****************************************************************

Jason2866, 2022-11-23:
Looking in examples may help too ;-)
https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-wifiscan