platformio/platform-espressif32 #25: undefined reference to app_main

FlashPackets, 2017-06-20:
Looking forward, the best way to resolve this issue may be to provide a few examples where there is greater program complexity with multiple subdirectories that approach a real world problem environment. In this case the working example is Pcbreflux's program ESP32_mqtts_gpio. In this example there are three user defined programs. mqtt_subscribe_main.c that has app_main(), MQTTClient.c residing in the MQTTClient-C subdirectory, and 10 or so .c files in MQTTPacket subdirectory from which I've never seen a compiler error. Within each of the subdirectories there is an empty component.mk file, at the same level as the .c file. One level up at main is another copy of an empty component.mk.
In this configuration Build all will generate the following:/Users/flashpackets/esp/workspace/ESP32_mqtts_gpio/main/./mqtt_subscribe_main.c:27:24: fatal error: MQTTClient.h: No such file or directory
compilation terminated.
If the top level make file is renamed component.mk_ Build all produces:/Users/flashpackets/esp/esp-idf/components/esp32/./cpu_start.c:305: undefined reference to `app_main'

The programs are being developed on a Mac. Not that it should matter.
As it has come up in a previous post regarding this issue the results of running make V=1 are attached. 
[makeV1.TXT](https://github.com/platformio/platform-espressif32/files/1086755/makeV1.TXT)

ivankravets, 2017-12-13:
Could you provide test project to reproduce this issue?

ivankravets, 2018-02-08:
Please reopen if you still need help

closed 2018-02-08 by ivankravets

kemaldede, 2018-05-25:
Did you solved the problem?

ivankravets, 2018-05-25:
@kemaldede how to reproduce this issue? Could you provide a demo project?