platformio/platform-espressif32 #28: Ignoring commented out "#include Arduino.h" when compiling

danielcrowley, 2017-07-08:
I'm debugging a bigger problem that I've whittled down to this example
Description:
Compiler ignores commenting in the line `//#include "Arduino.h"` and tries to include it anyway

Reproduce:
 - Take the espidf_httprequest example code (should work with any)
 - Add a header file in a private library (in this case lib/Common/Common.h)
 - Add single include `#include "Arduino.h"` (this fails as in #24, although it says it can't find common.h)
 - Change the line to `//#include "Arduino.h"` (This should still fail)
 - Only compiles again when the line is deleted.

[image]

ivankravets, 2017-07-24:
This is a bug of PIO Core. Please move this issue to https://github.com/platformio/platformio-core/issues

ivankravets, 2018-02-08:
Could you reproduce this issue with the latest version?

closed 2018-05-25 by ivankravets