platformio/platform-espressif32 #30: Upload fails when upload_flags debug is set

baldram, 2017-07-15:
What kind of issue is this?

- [x] Bug report. If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:
Linux Mint 18 Sarah
Linux 4.4.0-83-generic x86_64

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.4.0

### Description of problem
Unable to upload the compiled firmware with debug option enabled as described in [specification](http://docs.platformio.org/en/latest/platforms/espressif32.html#authentication-and-upload-options). This prevents me of using `ESP_LOGD` (part of Logging library `esp_log.h`) to display debug messages.

#### Steps to Reproduce

1. Please create hello world project using "espidf" framework for "espressif32" platform ("esp32dev" board).
2. In `platformio.ini` pleae add `upload_flags = --debug`. 
3. Connect the device via USB and try to upload the compiled firmware with command:
```
platformio run --target upload
```

### Actual Results
It ends up with error like below:
```console
Auto-detected: /dev/ttyUSB0
Uploading .pioenvs/esp32dev/firmware.bin
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
[--flash_mode {qio,qout,dio,dout}]
[--flash_size FLASH_SIZE] [--ucIsHspi]
[--ucIsLegacy] [--no-progress] [--verify]
[--compress | --no-compress]
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Must be pairs of an address and the binary filename to write there
*** [upload] Error 2
```
When the `upload_flags` in line platformio.ini is commented, everything is fine and it ends up with "success" message. 

### Expected Results
It should ends up with the "success" message.
```
================================================================== [SUCCESS] Took 44.22 seconds ==================================================================
```

**The content of `platformio.ini`:**
```ini
[env:esp32dev]
platform = espressif32
framework = espidf
board = esp32dev
upload_flags = --debug
```

**Source file to reproduce issue:**

As the source code please use the example taken from [here](https://github.com/platformio/platformio-examples/blob/develop/espressif/esp32-espidf-hello-world/src/main.c).
The goal was to replace `printf` with `ESP_LOGD` but then I need to have a possibility to enable debug mode as by default messages printed by the `ESP_LOGD` are not displayed.

### Additional info

I tried to study the specification and the [Authentication and upload options](http://docs.platformio.org/en/latest/platforms/espressif32.html#authentication-and-upload-options) section. I followed the instructions. If I understood something wrong, could you please clarify the spec?
So in case it this is not a but, could you please describe how to successfully enable debug mode in my project? 
If this is a bug, any temporary workaround would be also very welcome.

Thanks!

baldram, 2017-07-23:
I think the issue I described below regarding error related to upload_flags is because --debug option applies to OTA uploader, not the standard USB one.

Then the question is how should I define the log level via `platformio.ini` file? 
How about the correct approach then? 

As there is no error. Please close this issue then.

valeros, 2017-07-24:
Hi, @baldram!
`upload_flags = --debug` option has nothing to do with debug output in compiled firmware. This option can be only applied to an upload tool. In your case, you just need to increase the log verbosity by using `LOG_LOCAL_LEVEL` macro.  Could you please try something like this:
```
build_flags = -D LOG_LOCAL_LEVEL=ESP_LOG_DEBUG
```

baldram, 2017-07-28:
Yes, I've just wanted to update this by providing an example with `build_flags` what works very well!
An example `platformio.ini` to mute messages.
```
[env:esp32dev]
platform = espressif32
framework = espidf
board = esp32dev
build_flags = -DLOG_LOCAL_LEVEL=ESP_LOG_NONE
```
I'm able to set any level according to specification: https://esp-idf.readthedocs.io/en/v1.0/api/log.html#logging-library .

Thanks, as mentioned earlier, please close this incident.

closed 2017-07-28 by ivankravets