platformio/platform-espressif32 #33: Last stage version problem

fadsalah, 2017-08-01:
Hi, 

Version: 6552413 of the Espressif 32 Stage platform is causing the esp to be stuck in a reset loop whatever the code :  (TG1WDT_SYS_RESET)

Compiling with Arduino IDE using the same version works fine.

maxcww, 2017-08-16:
Hi, when will this be fixed? I have to use the bluetooth functionality, but it still bootloops.

ivankravets, 2017-08-16:
Resolved in https://github.com/platformio/platform-espressif32/releases/tag/v0.9.0

Please `pio update`

maxcww, 2017-08-17:
Hey, I have updated it like you said and in the platformio.ini file i have platform = espressif32_stage. But after building it, the esp32 still bootloops.

ivankravets, 2017-08-17:
Please share here your `platformio.ini`.

maxcww, 2017-08-17:
This is my platformio.ini file
`[env:esp32]`
`platform = espressif32_stage`
`framework = arduino`
`board = nodemcu-32s`
`lib_deps = ArduinoJson`
`           DHT sensor library`
`           TFT_eSPI`
`monitor_baud = 115200`
`upload_speed = 460800`

So in the terminal I have done the following: `platformio platform install https://github.com/platformio/platform-espressif32.git#feature/stage`

When I use the platform espressif32 my bluetooth files do not compile because of not defined variables, while using the platform espresif32_stage it compiles, but the esp goes into a bootloop.

ivankravets, 2017-08-17:
Could you try to add to `platformio.ini` 
```
[env:esp32]
platform = espressif32_stage
framework = arduino
board = nodemcu-32s
lib_deps = ArduinoJson
DHT sensor library
TFT_eSPI
monitor_baud = 115200
upload_speed = 460800
lib_archive = false
```

Does it work now?

maxcww, 2017-08-17:
Nope :(. I keep getting the following message in the Serial monitor:

`rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)`
`configsip: 0, SPIWP:0xee`
`clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00`
`mode:DIO, clock div:1`
`load:0x3fff0010,len:4`
`load:0x3fff0014,len:588`
`load:0x40078000,len:0`

`load:0x40078000,len:10472`
`entry 0x40078a28`

valeros, 2017-08-17:
Hi @maxcww !
Could you please try to completely erase flash memory and then flash your board again?
Go to `~/.platformio/packages/tool-esptoolpy` folder (on Windows `C:\Users\YOUR_USER\.platformio\packages\tool-esptoolpy`)and run next command:
```
python esptool.py --chip esp32 --port "INSERT_ESP_PORT" --baud 115200 --before "default_reset" --after "hard_reset" erase_flash
```

maxcww, 2017-08-17:
Hello, I have done that and then restarted atom and uploaded everything again, but it still gives me the same messages over and over.

jukebox, 2017-08-30:
> rst:0x8 (TG1WDT_SYS_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0010,len:4
> load:0x3fff0014,len:588
> load:0x40078000,len:0
> load:0x40078000,len:9768
> entry 0x400789b8

same issue here, with 

[env:esp32-evb]
platform = espressif32_stage
board = esp32-evb
framework = arduino

how can i help you ?

PheiPheiPhei, 2017-09-02:
Same issue. Latest arduino-esp32 version compiles and works fine with the Arduino IDE, but it gets stuck in Platformio. 

Platformio.ini 

> [env:esp32dev]
> platform = espressif32_stage
> board = esp32doit-devkit-v1
> framework = arduino
> upload_port = COM4
> upload_speed = 921600
> lib_deps =
>   https://github.com/bbx10/WebServer_tng.git
>   https://github.com/bbx10/DNSServer_tng.git
>   https://github.com/bbx10/Hash_tng.git

Error messages:

> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> 
> load:0x3fff0010,len:4
> load:0x3fff0014,len:588
> load:0x40078000,len:0
> load:0x40078000,len:9768
> entry 0x400789b8
> ets Jun  8 2016 00:22:57
> 
> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0010,len:4
> load:0x3fff0014,len:588
> load:0x40078000,len:0
> load:0x40078000,len:9768
> entry 0x400789b8

[Arduino IDE build log]( https://gist.github.com/PheiPheiPhei/0f33f5d0c4c6c3cc2e37cf54e4a134b4)

[Platformio build log](https://gist.github.com/PheiPheiPhei/be644f060cc9260c22355779165f62f2)

ivankravets, 2017-09-02:
@valeros please take a look at all changes since Aug 1, 2017. It seems that not all changes were applied to PIO Build script. 

Please make PR to https://github.com/espressif/arduino-esp32 and @me-no-dev will merge.

me-no-dev, 2017-09-03:
@ivankravets I wonder what could that be...

cranphin, 2017-09-04:
Same here, working on arduino-ide, not on platformio. Is there a manual fix I can do for this? It's not very clear from the comments what causes the issue: 'It seems that not all changes were applied to PIO Build script.', are those changes anywhere to be found?

cranphin, 2017-09-04:
Well, after comparing build logs till I was starting to get dizzy, adding `"-u", "ld_include_panic_highint_hdl"` to platformio-build.py seems to help:
```
    LINKFLAGS=[
        "-Wl,-EL",
        "-T", "esp32.common.ld",
        "-T", "esp32.rom.ld",
        "-T", "esp32.peripherals.ld",
        "-u", "ld_include_panic_highint_hdl"
    ],
```

me-no-dev, 2017-09-04:
That must be it!

me-no-dev, 2017-09-04:
Fixed!

PheiPheiPhei, 2017-09-04:
Thanks guys!

ivankravets, 2017-09-04:
Thanks all!!!

closed 2017-09-04 by ivankravets