platformio/platform-espressif32 #34: ESP-WROVER-KIT and SparkFun BME280 library compilation failure

ButchAnton, 2017-08-10:
When trying to compile the simplest of test programs with PIO and the SparkFun BME280 library, I receive the following errors:

```
$ cat main.cpp
#include <stdint.h>
#include <Wire.h>
#include <SPI.h>
#include "SparkFunBME280.h"

BME280 bme;

void setup() {

}

void loop() {

}
$ pio run
[Thu Aug 10 13:53:22 2017] Processing esp-wrover-kit (platform: espressif32; board: esp-wrover-kit; framework: arduino)
--------------------------------------------------------------------------------
Collected 15 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <SPI> v1.0 (/Users/i831533/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|-- <SparkFun BME280> v1.1.0 (/Users/i831533/src/ESP32/wrover/bme280_test/.piolibdeps/SparkFun BME280_ID684)
|   |-- <SPI> v1.0 (/Users/i831533/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|   |-- <Wire> v1.0 (/Users/i831533/.platformio/packages/framework-arduinoespressif32/libraries/Wire)
|-- <Wire> v1.0 (/Users/i831533/.platformio/packages/framework-arduinoespressif32/libraries/Wire)
xtensa-esp32-elf-ar rcs .pioenvs/esp-wrover-kit/libFrameworkArduino.a .pioenvs/esp-wrover-kit/FrameworkArduino/Esp.o .pioenvs/esp-wrover-kit/FrameworkArduino/HardwareSerial.o .pioenvs/esp-wrover-kit/FrameworkArduino/IPAddress.o .pioenvs/esp-wrover-kit/FrameworkArduino/IPv6Address.o .pioenvs/esp-wrover-kit/FrameworkArduino/MD5Builder.o .pioenvs/esp-wrover-kit/FrameworkArduino/Print.o .pioenvs/esp-wrover-kit/FrameworkArduino/Stream.o .pioenvs/esp-wrover-kit/FrameworkArduino/StreamString.o .pioenvs/esp-wrover-kit/FrameworkArduino/WMath.o .pioenvs/esp-wrover-kit/FrameworkArduino/WString.o .pioenvs/esp-wrover-kit/FrameworkArduino/base64.o .pioenvs/esp-wrover-kit/FrameworkArduino/cbuf.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-adc.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-bt.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-dac.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-gpio.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-i2c.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-ledc.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-matrix.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-misc.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-sigmadelta.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-spi.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-time.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-timer.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-touch.o .pioenvs/esp-wrover-kit/FrameworkArduino/esp32-hal-uart.o .pioenvs/esp-wrover-kit/FrameworkArduino/libb64/cdecode.o .pioenvs/esp-wrover-kit/FrameworkArduino/libb64/cencode.o .pioenvs/esp-wrover-kit/FrameworkArduino/main.o .pioenvs/esp-wrover-kit/FrameworkArduino/stdlib_noniso.o .pioenvs/esp-wrover-kit/FrameworkArduino/wiring_pulse.o .pioenvs/esp-wrover-kit/FrameworkArduino/wiring_shift.o
xtensa-esp32-elf-ar rcs .pioenvs/esp-wrover-kit/lib/libWire.a .pioenvs/esp-wrover-kit/lib/Wire/Wire.o
xtensa-esp32-elf-ranlib .pioenvs/esp-wrover-kit/lib/libWire.a
xtensa-esp32-elf-g++ -o ".pioenvs/esp-wrover-kit/lib/SparkFun BME280_ID684/SparkFunBME280.o" -c -fno-rtti -fno-exceptions -std=gnu++11 -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -Os -g3 -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -DARDUINO=10610 -DARDUINO_ARCH_ESP32 -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPLATFORMIO=30500 -DESP32_DEV -DARDUINO_ESP32_DEV -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/config -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bluedroid -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/app_update -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bootloader_support -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bt -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/driver -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp32 -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/ethernet -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/fatfs -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/jsmn -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/log -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/mdns -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/mbedtls -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/mbedtls_port -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/nvs_flash -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/openssl -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/soc -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/spi_flash -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/sdmmc -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/tcpip_adapter -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/ulp -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/vfs -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/wear_levelling -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/xtensa-debug-module -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/coap -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/wpa_supplicant -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/expat -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/json -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/nghttp -I/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip -I/Users/butch/.platformio/packages/framework-arduinoespressif32/cores/esp32 -I/Users/butch/.platformio/packages/framework-arduinoespressif32/variants/esp32 -I/Users/i831533/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src -I/Users/i831533/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src "-I.piolibdeps/SparkFun BME280_ID684/src" ".piolibdeps/SparkFun BME280_ID684/src/SparkFunBME280.cpp"
xtensa-esp32-elf-ranlib .pioenvs/esp-wrover-kit/libFrameworkArduino.a
.piolibdeps/SparkFun BME280_ID684/src/SparkFunBME280.cpp: In member function 'uint8_t BME280::begin()':
.piolibdeps/SparkFun BME280_ID684/src/SparkFunBME280.cpp:82:19: error: 'MSBFIRST' was not declared in this scope
SPI.setBitOrder(MSBFIRST);
^
*** [.pioenvs/esp-wrover-kit/lib/SparkFun BME280_ID684/SparkFunBME280.o] Error 1
```

valeros, 2017-08-11:
Hi @ButchAnton !
Just tried your example in Arduino IDE and there is the same error:
[image]
Am I missing something?
Anyway, I see that `MSBFIRST` is defined in `Arduino.h`, so as a workaround you can simply add 
`#include <Arduino.h>` line to `SparkFunBME280.cpp` file.

ButchAnton, 2017-08-11:
@valeros I do not get any error in the Arduino IDE when compiling.  I am using the hourly build 1.8.4 from 2017/08/10@11:12.
```
#include <stdint.h>
#include <Wire.h>
#include <SPI.h>
#include "SparkFunBME280.h"

BME280 bme;

void setup() {

}

void loop() {

}
```

```
/Applications/Arduino-nightly.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino-nightly.app/Contents/Java/hardware -hardware /Users/i831533/Library/Arduino15/packages -hardware /Users/butch/Documents/Arduino/hardware -tools /Applications/Arduino-nightly.app/Contents/Java/tools-builder -tools /Applications/Arduino-nightly.app/Contents/Java/hardware/tools/avr -tools /Users/i831533/Library/Arduino15/packages -built-in-libraries /Applications/Arduino-nightly.app/Contents/Java/libraries -libraries /Users/butch/Documents/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=40,UploadSpeed=921600,DebugLevel=none -ide-version=10804 -build-path /var/folders/7g/mzjkvdxn5jb7dj9wty16tl_mxyly6b/T/arduino_build_560065 -warnings=default -prefs=build.warn_data_percentage=75 -verbose /Users/i831533/src/ESP32/wrover/simple_test_arduino/simple_test_arduino.ino
/Applications/Arduino-nightly.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino-nightly.app/Contents/Java/hardware -hardware /Users/i831533/Library/Arduino15/packages -hardware /Users/butch/Documents/Arduino/hardware -tools /Applications/Arduino-nightly.app/Contents/Java/tools-builder -tools /Applications/Arduino-nightly.app/Contents/Java/hardware/tools/avr -tools /Users/i831533/Library/Arduino15/packages -built-in-libraries /Applications/Arduino-nightly.app/Contents/Java/libraries -libraries /Users/butch/Documents/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=40,UploadSpeed=921600,DebugLevel=none -ide-version=10804 -build-path /var/folders/7g/mzjkvdxn5jb7dj9wty16tl_mxyly6b/T/arduino_build_560065 -warnings=default -prefs=build.warn_data_percentage=75 -verbose /Users/i831533/src/ESP32/wrover/simple_test_arduino/simple_test_arduino.ino
Using board 'esp32' from platform in folder: /Users/butch/Documents/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /Users/butch/Documents/Arduino/hardware/espressif/esp32
Detecting libraries used...
"/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/heap" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/soc" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/coap" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/expat" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/json" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp3 [... cut ...]
Using library Wire at version 1.0 in folder: /Users/butch/Documents/Arduino/hardware/espressif/esp32/libraries/Wire 
Using library SPI at version 1.0 in folder: /Users/butch/Documents/Arduino/hardware/espressif/esp32/libraries/SPI 
Sketch uses 276111 bytes (21%) of program storage space. Maximum is 1310720 bytes.
Global variables use 26780 bytes (9%) of dynamic memory, leaving 268132 bytes for local variables. Maximum is 294912 bytes.
```
I have chosen the ESP32 Dev Module for the board and changed the flash speed to 40MHz.

valeros, 2017-08-11:
@ButchAnton 
hm, I don't see compiled BME280 library in your log. Could you please build [this example](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/master/examples/CSV_Output/CSV_Output.ino)? does it work?

ButchAnton, 2017-08-11:
Yes, it builds correctly:
```
/Applications/Arduino-nightly.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino-nightly.app/Contents/Java/hardware -hardware /Users/i831533/Library/Arduino15/packages -hardware /Users/butch/Documents/Arduino/hardware -tools /Applications/Arduino-nightly.app/Contents/Java/tools-builder -tools /Applications/Arduino-nightly.app/Contents/Java/hardware/tools/avr -tools /Users/i831533/Library/Arduino15/packages -built-in-libraries /Applications/Arduino-nightly.app/Contents/Java/libraries -libraries /Users/butch/Documents/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=40,UploadSpeed=921600,DebugLevel=none -ide-version=10804 -build-path /var/folders/7g/mzjkvdxn5jb7dj9wty16tl_mxyly6b/T/arduino_build_560065 -warnings=default -prefs=build.warn_data_percentage=75 -verbose /Users/i831533/src/ESP32/wrover/simple_test_arduino/simple_test_arduino.ino
/Applications/Arduino-nightly.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino-nightly.app/Contents/Java/hardware -hardware /Users/i831533/Library/Arduino15/packages -hardware /Users/butch/Documents/Arduino/hardware -tools /Applications/Arduino-nightly.app/Contents/Java/tools-builder -tools /Applications/Arduino-nightly.app/Contents/Java/hardware/tools/avr -tools /Users/i831533/Library/Arduino15/packages -built-in-libraries /Applications/Arduino-nightly.app/Contents/Java/libraries -libraries /Users/butch/Documents/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=40,UploadSpeed=921600,DebugLevel=none -ide-version=10804 -build-path /var/folders/7g/mzjkvdxn5jb7dj9wty16tl_mxyly6b/T/arduino_build_560065 -warnings=default -prefs=build.warn_data_percentage=75 -verbose /Users/i831533/src/ESP32/wrover/simple_test_arduino/simple_test_arduino.ino
Using board 'esp32' from platform in folder: /Users/butch/Documents/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /Users/butch/Documents/Arduino/hardware/espressif/esp32
Detecting libraries used...
"/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/heap" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/soc" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/coap" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/expat" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/json" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/Users/butch/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable - [... cut ...] 560065/simple_test_arduino.ino.elf"
esptool.py v2.1-beta1
Using library Wire at version 1.0 in folder: /Users/butch/Documents/Arduino/hardware/espressif/esp32/libraries/Wire 
Using library SPI at version 1.0 in folder: /Users/butch/Documents/Arduino/hardware/espressif/esp32/libraries/SPI 
Sketch uses 292811 bytes (22%) of program storage space. Maximum is 1310720 bytes.
Global variables use 27256 bytes (9%) of dynamic memory, leaving 267656 bytes for local variables. Maximum is 294912 bytes.
```

valeros, 2017-08-11:
@ButchAnton 
Looks like you haven't installed BME280 library properly, you just copied source files to your project. Could you please try to install this library using Library Manager in Arduino IDE?

ButchAnton, 2017-08-11:
When installing the library via the Arduino IDE Library Manager, the code fails to build with the same error as in PIO:
```
/Users/butch/Documents/Arduino/libraries/SparkFun_BME280/src/SparkFunBME280.cpp: In member function 'uint8_t BME280::begin()':
/Users/butch/Documents/Arduino/libraries/SparkFun_BME280/src/SparkFunBME280.cpp:82:19: error: 'MSBFIRST' was not declared in this scope
   SPI.setBitOrder(MSBFIRST);
```

valeros, 2017-08-11:
@ButchAnton So, to solve this problem you have two ways:
* Add missing defines to build flags: `build_flags = -DLSBFIRST=0 -DMSBFIRST=1`
* Open `SparkFunBME280.cpp` and add line `#include <Arduino.h>` somewhere at the beginning of the file.

ButchAnton, 2017-08-11:
This issue is fixed in the version on GitHub:

git@github.com:sparkfun/SparkFun_BME280_Arduino_Library.git

closed 2017-08-11 by ButchAnton

1technophile, 2018-01-05:
This issue is reappearing for me (same compilation error "error: 'MSBFIRST' was not declared in this scope") with the last update of ESP32 arduino dev
And corrected if I add #include <Arduino.h> into the SparkFunBME280.cpp