platformio/platform-espressif32 #35: toolchain-xtensa32

pmaduarte, 2017-08-11:
Hi,

I'm having this error on "run".

[Fri Aug 11 17:48:13 2017] Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)

PackageManager: Installing toolchain-xtensa32 @ ~1.50200.0
Error: Could not find a version that satisfies the requirement '~1.50200.0' for your system 'linux_armv7l'

How do i solve this?

ivankravets, 2017-08-16:
@igrr do you plan to release ARMv6+ version of ESP32 toolchain??

arunderwood, 2018-02-09:
I am also encountering this error.  Is this a bug, or a case of the toolchain-xtensa32 not being available for ARM platforms (ie. It is not supported to flash an ESP32 from a RaspberryPi)?

tsathishkumar, 2018-05-21:
Any plan for this enhancement yet? We are planning to use a RaspberryPi as a CI machine and we are stuck with installing `toolchain-xtensa32`.

tsathishkumar, 2018-05-23:
@ivankravets Is there a way to configure platformio to download the xtensa32 toolchain from custom link?

tsathishkumar, 2018-05-25:
@ivankravets sorry, I forgot to publish the changes. Now it is available [here](https://bintray.com/tsathishkumar/toolchain-xtensa32/toolchain-xtensa32/2.50200.80#files)

ivankravets, 2018-05-25:
Thanks! I've just deployed to our registry.

closed 2018-05-25 by ivankravets