platformio/platform-espressif32 #38: MQTT Example, several correctable build errors and an execution fail, needs correction

frank3141, 2017-08-24:
Downloaded the Tuan MQTT example and the dependency:
1. esp32-mqtt
2. espmqtt
Restructured for Platformio, all files in src.
Build errors centered on incomplete removal of reconnect cxall back and a variable BUIDTIME that is undefined.  Removed the reconnect references and the reference to BUIDTIME and obtained a clean compile after adding defines for SSID and PASSWORD variables.  Did this by #include USER config file and corected varable names.
Using an ESP32thing from Sparkfun.  This example worked prior to the recent updates.  Now it fails and never gets to the MQTT part.
Log below:

> Executing task: platformio.exe device monitor <

--- Miniterm on COM14  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:4408
load:0x40078000,len:11072
load:0x40080000,len:252
entry 0x40080034
␛[0;32mI (29) boot: ESP-IDF 2.210.0 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 13:16:38␛[0m
␛[0;32mI (62) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (62) boot: SPI Speed      : 80MHz␛[0m
␛[0;32mI (62) boot: SPI Mode       : DIO␛[0m
␛[0;32mI (74) boot: SPI Flash Size : 4MB␛[0m
␛[0;32mI (86) boot: Partition Table:␛[0m
␛[0;32mI (97) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (120) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (143) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (166) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (190) boot: End of partition table␛[0m
␛[0;32mI (202) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (220) boot: Loading app partition at offset 00010000␛[0m
␛[0;32mI (1085) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) ␛[0m
␛[0;32mI (1086) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x1812c ( 98604) map␛[0m
␛[0;32mI (1102) boot: segment 2: paddr=0x0003813c vaddr=0x3ffc0000 size=0x02ca8 ( 11432) load␛[0m
␛[0;32mI (1131) boot: segment 3: paddr=0x0003adec vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
␛[0;32mI (1156) boot: segment 4: paddr=0x0003b1f4 vaddr=0x40080400 size=0x12c14 ( 76820) load␛[0m
␛[0;32mI (1204) boot: segment 5: paddr=0x0004de10 vaddr=0x400c0000 size=0x00000 (     0) load␛[0m
␛[0;32mI (1208) boot: segment 6: paddr=0x0004de18 vaddr=0x00000000 size=0x021f0 (  8688) ␛[0m
␛[0;32mI (1233) boot: segment 7: paddr=0x00050010 vaddr=0x400d0018 size=0x6a684 (435844) map␛[0m
␛[0;32mI (1260) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (1271) cpu_start: Starting app cpu, entry point is 0x40080e74␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (1304) heap_alloc_caps: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (1326) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM␛[0m
␛[0;32mI (1347) heap_alloc_caps: At 3FFC81F0 len 00017E10 (95 KiB): DRAM␛[0m
␛[0;32mI (1368) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (1389) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (1411) heap_alloc_caps: At 40093014 len 0000CFEC (51 KiB): IRAM␛[0m
␛[0;32mI (1432) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (1487) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (1489) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (1489) MQTT_SAMPLE: [APP] Startup..␛[0m
␛[0;32mI (1489) MQTT_SAMPLE: [APP] Free memory: 210708 bytes␛[0m
␛[0;32mI (1499) MQTT_SAMPLE: [APP] SDK version: master␛[0m
I (1529) wifi: wifi firmware version: 407bb27
I (1529) wifi: config NVS flash: enabled
I (1529) wifi: config nano formating: disabled
␛[0;32mI (1529) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
␛[0;32mI (1539) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
I (1579) wifi: Init dynamic tx buffer num: 32
I (1579) wifi: Init dynamic rx buffer num: 32
I (1579) wifi: wifi driver task: 3ffcbec4, prio:23, stack:4096
I (1579) wifi: Init static rx buffer num: 10
I (1589) wifi: Init dynamic rx buffer num: 32
I (1589) wifi: Init rx ampdu len mblock:7
I (1589) wifi: Init lldesc rx ampdu entry mblock:4
I (1599) wifi: wifi power manager task: 0x3ffd4fbc prio: 21 stack: 2560

So ffar so good, I think, to here

␛[0;32mI (1599) MQTT_SAMPLE: start the WIFI SSID:[IOTDeva] password:[******]␛[0m

SSID and Password are coreect and work with other applications

I (1609) wifi: wifi timer task: 3ffd6024, prio:22, stack:3584
␛[0;32mI (1649) phy: phy_version: 355.1, 59464c5, Jun 14 2017, 20:25:06, 1, 0␛[0m
I (1649) wifi: mode : sta (24:0a:c4:05:cf:7c)
I (1889) wifi: n:2 0, o:1 0, ap:255 255, sta:2 0, prof:1
I (2869) wifi: state: init -> auth (b0)
I (3869) wifi: state: auth -> init (2)
I (3869) wifi: n:2 0, o:2 0, ap:255 255, sta:2 0, prof:1
I (3989) wifi: n:2 0, o:2 0, ap:255 255, sta:2 0, prof:1
I (3989) wifi: state: init -> auth (b0)
I (4989) wifi: state: auth -> init (2)
I (4999) wifi: n:2 0, o:2 0, ap:255 255, sta:2 0, prof:1
I (5119) wifi: n:2 0, o:2 0, ap:255 255, sta:2 0, prof:1
I (5119) wifi: state: init -> auth (b0)
I (6119) wifi: state: auth -> init (2)
I (6119) wifi: n:2 0, o:2 0, ap:255 255, sta:2 0, prof:1
I (6239) wifi: n:2 0, o:2 0, ap:255 255, sta:2 0, prof:1
I (6239) wifi: state: init -> auth (b0)
I (7239) wifi: state: auth -> init (2)
I (7239) wifi: n:2 0, o:2 0, ap:255 255, sta:2 0, prof:1
I (7359) wifi: n:2 0, o:2 0, ap:255 255, sta:2 0, prof:1
I (7359) wifi: state: init -> auth (b0)
This latter part repeats for ever and no MQTT subscription occurs either with mosquitto or my local broker 
Any help appreciated.
Thanks
Frank

ivankravets, 2017-12-13:
Could you reproduce this issue with the latest version?

beegee-tokyo, 2017-12-18:
@frank3141. I am using [arduino-mqtt](https://github.com/256dpi/arduino-mqtt) without problems on my ESP32. Maybe you can try that client. Works reliable for me on ESP32 and ESP8266.

frank3141, 2017-12-18:
Ivan,

Finally got it to work.  Had to fix the code issues originally cited and 
then modify sdkconfig.h  as make config doesn't appear to be implemented 
in platformiio.

Frank

ivankravets, 2017-12-18:
Thanks for the report!

closed 2017-12-18 by ivankravets