platformio/platform-espressif32 #41: Linking error with bluedroid

DavidBertet, 2017-09-05:
I'm trying to use BLE with bluedroid methods but it doesn't work

The configuration seems good
```
#define CONFIG_BT_ENABLED 1
#define CONFIG_BLUEDROID_ENABLED 1
```

My code is pretty simple
```
#include "esp_bt_main.h"

void app_main() {
    esp_bluedroid_init();
}
```

The project compiles but can't be linked :
```
Compiling .pioenvs/esp32dev/src/main.o
Linking .pioenvs/esp32dev/firmware.elf

.pioenvs/esp32dev/src/main.o:(.literal.app_main+0x0): undefined reference to `esp_bluedroid_init'
.pioenvs/esp32dev/src/main.o: In function `app_main':
```

The bluedroid dependency is never compiled but I not finding why.

Any idea ?

My platformio.ini is fairly basic
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
```

valeros, 2017-09-05:
Hi, @DavidBertet!
Many thanks for pointing this out. Could you please try to install the `espressif32` platform from dev branch and compile your project again?
```
pio platform uninstall espressif32
pio platform install https://github.com/platformio/platform-espressif32.git
```

valeros, 2017-09-06:
You are right about the conflict with `config.h` files and probably the best workaround is to rename your configuration file.

DavidBertet, 2017-09-09:
Ok, thanks again @valeros, everything works fine !

closed 2017-09-09 by DavidBertet