platformio/platform-espressif32 #44: [Feature] Add support for esf-idf project

fanthos, 2017-09-11:
I am trying to build https://github.com/MrBuddyCasino/ESP32_MP3_Decoder .
That project using component.mk, component folder in project and Kconfig, so that project is not able to build on PlatformIO without lots of modification.
I'm also checking examples from official esp32 repo, the source files are located on folder `main`.
Are there any plans to support esf-idf project directly or import?

Another problem is that PlatformIO does not have Kconfig support. After I run `make menuconfig` I got sdkconfig, I have to manually to convert it to sdkconfig.h if I want to build the project using PlatformIO. I hope there is easier way to do that.

huming2207, 2017-09-15:
Agree. But the problem is that ESP-IDF needs to configure with KBuild (the ```make menuconfig```) unless some extra workarounds are implemented. For example, Arduino-ESP32 used precompiled binaries from ESP-IDF.

fanthos, 2017-09-15:
The KBuild file can be processed by a library like https://github.com/ulfalizer/Kconfiglib , but still lots of work need to be done.

Gouthamm4G, 2018-04-18:
@fanthos @huming2207 But right now, just supporting the folder structure of the esp-idf project layout and configuring them separately in the terminal would still be a quick viable workaround. It wouldn't need much changes to the existing system. Just add a few extra targets to the `esp-idf.py` file within the platform folder as per your requirements and you are good to go.

ivankravets, 2018-05-25:
Duplicate of https://github.com/platformio/platform-espressif32/issues/20

closed 2018-05-25 by ivankravets