platformio/platform-espressif32 #46: PlatformIO does not support Pumbaa on ESP32 Devkit, although Pumbaa says they do support it.

MrDadaGuy, 2017-09-20:
`
[Tue Sep 19 10:24:49 2017] Processing esp32dev (platform: espressif32; board: esp32dev; framework: pumbaa)

Verbose mode can be enabled via -v, --verbose option
Error: This board doesn't support pumbaa framework!
======================================================== [ERROR] Took 0.48 seconds ========================================================
The terminal process terminated with exit code: 1`

Here is the information on Pumbaa site:

http://pumbaa.readthedocs.io/en/latest/boards/esp32_devkitc.html

crashGoBoom, 2017-10-29:
I was having the same issue @MrDadaGuy but I was able to get around it by overriding the nano32.json config since thats one of the few the platformio simba framework implementation supports right now. I just created the following in `~/.platformio/boards/nano32.json`

```
{
  "build": {
    "core": "esp32",
    "extra_flags": "-DESP32_DEV -DARDUINO_ESP32_DEV",
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dio",
    "ldscript": "esp32_out.ld",
    "mcu": "esp32",
    "variant": "esp32dev"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "frameworks":  [
    "arduino",
    "espidf",
    "simba",
    "pumbaa"
  ],
  "name": "Espressif ESP32 Dev Module - Custom",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 294912,
    "maximum_size": 1310720,
    "require_upload_port": true,
    "resetmethod": "nodemcu",
    "speed": 115200,
    "wait_for_upload_port": true
  },
  "url": "https://en.wikipedia.org/wiki/ESP32",
  "vendor": "Espressif"
}
```
And in your platformio.ini for your project:
```
[env:your-custom-board-to-support-simba-or-pumbaa]
platform = espressif32
board = nano32
framework = simba
monitor_baud = 115200
```
When creating a new project you should be able to find it using the name key in the json file too. Found all of this info here: http://docs.platformio.org/en/latest/platforms/creating_board.html

Hope this helps!

ivankravets, 2017-11-26:
The only NANO32 is supported. See https://raw.githubusercontent.com/eerimoq/pumbaa-releases/master/arduino/esp32/package_pumbaa_esp32_index.json

/cc @eerimoq

closed 2017-11-26 by ivankravets