platformio/platform-espressif32 #47: PlatformIO UPLOAD button does only build, but no upload to device

beegee-tokyo, 2017-09-22:
# Environment:
**PlatformIO** 
Version Home 0.3.0·Core 3.5.0a15

**Plattform:**
Espressif32 Version 0.10.0
Espressif32_stage Version bcfaefd

**Framework:**
Arduino

# How to recreate
Any ESP32 project (or example) will show the problem.
Select project
Press UPLOAD button (or select from Menu PlatformIO->Upload
==> Build starts
==> After build is finished nothing else happens, last message is:
```
Calculating size .pioenvs\esp32dev\firmware.elf
text       data     bss     dec     hex filename
415909    72348   27160  515417   7dd59 .pioenvs\esp32dev\firmware.elf
 [SUCCESS] Took 2.70 seconds
```

#Workaround
Open PlatformIO terminal and enter 
`platformio run --target upload`
to build and upload

YannikW, 2017-11-01:
I notices the same problem with ESP8266.

Try to add ```tagets = upload``` to you projects platformio.ini, this works for me.

[image]

beegee-tokyo, 2017-11-02:
@YannikW thanks for the tip, that works. But now all buttons try to upload. so not the final solution

ivankravets, 2017-11-02:
What is your OS? Atom or VSCode?

ivankravets, 2017-11-02:
If Atom, please open PlatformIO IDE Settings and Check “Use Development version of IDE”. Please wait for a while. Restart.

beegee-tokyo, 2017-11-02:
@ivankravets 
I switched to development version of IDE. After a while I got a message to restart PlatformIO.
After restart I created a new project outside of the default location.
Now pushing "CLEAN" gives me error:
```
PlatformIO: Failed to run a command: clean
Please make sure that "build" package is installed and activated.
```
Pushing "UPLOAD" gives me error:
```
PlatformIO: Failed to run a command: upload
Please make sure that "build" package is installed and activated.
```
Pushing "BUILD" gives me error :
```
PlatformIO: Failed to run a command: build
Please make sure that "build" package is installed and activated. 
```

Existing projects still works.
Creating new project in default folder still works.

Maybe I am doing something wrong. My usual steps:
1) Create empty directory in my project directory (which is e.g. B:\Projects\ESP32\Test)
2) From PlatformIO Home select `New Project`
3) Select board `Espressif ESP32 Dev Module`
4) Select Framework `Arduino`
5) Uncheck `Location: O Use default location`
6) Navigate to `B:\Projects\ESP32\Test` in the folder structure shown
7) Select folder `Test`
8) Click `Finish `and wait for message from PlatformIO

What I am doing right now to get around this problem is to create a project in the default project folder. Then I copy the folder that PlatformIO created to my prefered location and rename the folder to the prefered project name.
Then I add the folder manually with `File->Add Project Folder`. By doing this the buttons work as they should.
I tried to compare the project folder in the default location and the newly created one in my folder and cannot see any difference.

beegee-tokyo, 2017-11-03:
Tried to `pio upgrade`. First error message:
```
PS C:\Users\beegee> pio upgrade
Updating tool-pioplus                    @ 0.10.11        [0.10.13]
Uninstalling tool-pioplus @ 0.10.11:    Please manually remove file `C:\SPB_Data\.platformio\packages\tool-pioplu
s\pioplus.exe`
Error: Traceback (most recent call last):
  File "c:\users\beegee\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\beegee\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\beegee\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\beegee\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))

  File "c:\users\beegee\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\beegee\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\beegee\.platformio\penv\lib\site-packages\platformio\commands\upgrade.py", line 28, in cli
    update_core_packages(silent=True)
  File "c:\users\beegee\.platformio\penv\lib\site-packages\platformio\managers\core.py", line 91, in update_core_
packages
    pm.update(name, requirements, only_check=only_check)
  File "c:\users\beegee\.platformio\penv\lib\site-packages\platformio\managers\core.py", line 53, in update
    result = PackageManager.update(self, *args, **kwargs)
  File "c:\users\beegee\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 732, in update
    self.uninstall(pkg_dir, trigger_event=False)
  File "c:\users\beegee\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 673, in uninstal
l
    util.rmtree_(pkg_dir)
  File "c:\users\beegee\.platformio\penv\lib\site-packages\platformio\util.py", line 638, in rmtree_
    return rmtree(path, onerror=_onerror)

  File "C:\Python27\Lib\shutil.py", line 252, in rmtree
    onerror(os.remove, fullname, sys.exc_info())
  File "c:\users\beegee\.platformio\penv\lib\site-packages\platformio\util.py", line 636, in _onerror
    raise e
WindowsError: [Error 5] Access is denied: 'C:\\SPB_Data\\.platformio\\packages\\tool-pioplus\\pioplus.exe'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers

  https://github.com/platformio/platformio-core/issues

============================================================

```
As suggested in the error message, I manually removed `C:\SPB_Data\.platformio\packages\tool-pioplus\pioplus.exe` then tried `pio upgrade` again. Next error message:
```
PS C:\Users\beegee> pio upgrade
Please wait while upgrading PlatformIO ...
Error: ['c:\\users\\beegee\\.platformio\\penv\\scripts\\python.exe', '-m', 'pip', '--no-cache-dir', 'install', '-
-upgrade', 'https://github.com/platformio/platformio-core/archive/develop.zip']
Collecting https://github.com/platformio/platformio-core/archive/develop.zip
  Downloading https://github.com/platformio/platformio-core/archive/develop.zip (198kB)
Requirement already up-to-date: arrow<1 in c:\users\beegee\.platformio\penv\lib\site-packages (from platformio==3.5.0b1)
Requirement already up-to-date: bottle<0.13 in c:\users\beegee\.platformio\penv\lib\site-packages (from platformio==3.5.0b1)
Requirement already up-to-date: click<6,>=5 in c:\users\beegee\.platformio\penv\lib\site-packages (from platformio==3.5.0b1)
Requirement already up-to-date: colorama in c:\users\beegee\.platformio\penv\lib\site-packages (from platformio==3.5.0b1)
[... 58 lines cut ...]
* Upgrade using `pip install -U platformio`
* Try different installation/upgrading steps:
  http://docs.platformio.org/page/installation.html
```

So I did  `pip install -U platformio` which told me that I am already up to date?????

Next try of `pio upgrade` just said:
```
You're up-to-date!
`PlatformIO 3.5.0b1 is currently the newest version available.`
```

So I closed PlatformIO and deleted `$HOME_DIR%/.platformio/homestate.json`

Next start of PlatformIO the 'PlatformIO Home' tab is empty beside of a spinning busy icon.

I tried a 'pio update' and I saw a 'Updating contrib-piohome                 @ 0.3.1          [Fixed]' line.

But after restarting PlatformIO I still get the same empty 'PlatformIO Home' tab.

Next I went to 'PlatformIO->Settings' and saw that 'Use built-in PlatformIO Core' was unchecked. I changed it, but still getting an empty 'PlatformIO Home' tab.

Giving up at this point and will try a fresh clean install.    
Will give a feedback after the new installation is finished.

beegee-tokyo, 2017-11-03:
New installation finished, still having the problem with the empty 'PlatformIO Home' tab.
Trying 'pio home' in terminal throws
```
PS C:\Users\beegee> pio home
PlatformIO Plus (https://pioplus.com) v0.10.13
CorePackageManager: Installing contrib-piohome @ >=0.3.2,<2
Error: Detected unknown package 'contrib-piohome'
```

beegee-tokyo, 2017-11-03:
Solved that problem with downloading the **contrib-piohome** package manually and unpack it in packages folder. There is no version **0.3.2** available on [bintray](https://dl.bintray.com/platformio/dl-packages/), only **0.3.1**, I guess that might be the problem because in `core.py` it requests version 0.3.2.

But for the projects created outside of the default directory I still get the build errors like 
```
PlatformIO: Failed to run a command: clean
Please make sure that "build" package is installed and activated.
```

I checked that I am on developer IDE and latest core version.

ivankravets, 2017-11-03:
Fixed, just run `pio update`. Should be PIO Home 0.3.2

ivankravets, 2017-11-03:
1. Please be sure that you use PIO Core 3.5.0b1 and PIO Home 0.3.2
2. Close/remove all projects from Atom
3. Close Atom
4. Remove `%HOME_DIR%/.platformio/homestate.json` file
5. Start Atom and open project. Now, all should work.

closed 2017-11-03 by ivankravets

beegee-tokyo, 2017-11-04:
Confirmation
After another pio upgrade
```
PS C:\Users\beegee> pio upgrade
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 3.5.0b2
Release notes: http://docs.platformio.org/en/latest/history.html
```
The PIO Home screen shows
`Home 0.3.2·Core 3.5.0b1`

Creating a new project outside of the default location works now as expected. Clean, Build and Upload buttons work as excpected.

Thank you.

Jeremaiha-xmetix, 2020-08-02:
Had the very same issue, even when operating Arduino's native IDE.
Apparently, `reset` pin was connected (accidentally) and does prevent upload new code.