platformio/platform-espressif32 #50: Can't flash firmware on ESP32

albydnc, 2017-10-21:
###Issue:
Using default examples for esp32 boards, building is working, upload is not.
###Debug Error:
`Looking for upload port... Auto-detected: COM7 Uploading .pioenvs\esp32doit-devkit-v1\firmware.bin usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}] [--flash_mode {keep,qio,qout,dio,dout}] [--flash_size FLASH_SIZE] [--spi-connection SPI_CONNECTION] [--no-progress] [--verify] [--compress | --no-compress] <address> <filename> [<address> <filename> ...] esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\Alberto' *** [upload] Error 2 [ERROR] Took 8.11 seconds`

albydnc, 2017-11-06:
Not resolved.

ivankravets, 2017-11-26:
Thanks! Resolved in https://github.com/espressif/arduino-esp32/pull/881

closed 2017-11-26 by ivankravets

tablatronix, 2018-06-18:
hmm I started having this issue now

```
"c:\users\...\documents\projects\bin\sublimetext\data\packages\user\deviot\penv\scripts\python.exe" "C:\Users\...\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 --port "COM6" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect .pioenvs\esp32doit-devkit-v1\firmware.bin
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
[--flash_mode {keep,qio,qout,dio,dout}]
[--flash_size FLASH_SIZE]
[--spi-connection SPI_CONNECTION] [--no-progress]
[--verify] [--compress | --no-compress]
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address ".pioenvs\esp32doit-devkit-v1\firmware.bin" must be a number
*** [upload] Error 2
 [ERROR] Took 5.11 seconds 
```

ivankravets, 2018-06-18:
It seems that you have obsolete/broken ESP32 dev/ platform. Please open `$ pio home` and uninstall all ESP32 dev platforms. Start build project again.

tablatronix, 2018-06-18:
That did it, thanks

ivankravets, 2018-06-18:
Does it work now?

tablatronix, 2018-06-18:
Yes