platformio/platform-espressif32 #51: Can not upload files (and target uploadfs are missed in list in stage espressif32 platform)

skopanev, 2017-11-16:
Hey, thanks for a remarkable job you did, guys!

I can not upload my html file for captiva.
Previously I did it easily for esp8266 — via this menu: https://puu.sh/ymUG1.png with SPIFFS upload.
But now it shows error to me (and from console the same): 

> *** Do not know how to make File target `uploadfs' user/proj/uploadfs).  Stop.

I've tried with stage version as platform and there are some changes:
1/ Target UPLOAD SPIFFS is disappear from menu I showed above;
2/ Captiva not shown on devices. Back to "platform = espressif32" and everything is ok again;
3/ Can not compile SPIFFS which is from original github (on non-stage version of platform).

ivankravets, 2017-11-16:
Currently, we don't support SPIFFS for Espressif 32. Could you do that with Arduino IDE?

skopanev, 2017-11-16:
@ivankravets so in this case I can not upload files via uploadfs target, right?
And just wondering when you plan to add it : ) ?

closed 2017-11-17 by skopanev