platformio/platform-espressif32 #53: After last update platform = espressif32_stage is not working anymore

beegee-tokyo, 2017-11-26:
After the last update, if I have `platform = espressif32_stage` in my platformio.ini file I get the following error:
```
platformio run
[11/26/17 18:56:39] Processing esp32dev (build_flags: -O0; lib_deps: 1671; platform: espressif32_stage; board: esp32dev; framework: arduino; lib_ext
ra_dirs: B:\Projects\ESP32\ESP32Lib; upload_port: 192.168.0.106)
 
PlatformManager: Installing espressif32_stage
Error: Detected unknown package 'espressif32_stage'
```

`platform = https://github.com/platformio/platform-espressif32.git#feature/stage`
works as expected. 
Is this the expected behaviour???

ivankravets, 2017-11-26:
Yes, that is normal. We dropped support for `espressif32_stage` to avoid an issue with LIbrary Dependency Finder which compares libraries for platform compatibility. As result, if library as compatible with `espressif32` it was not compatible with `espressif32_stage`.

beegee-tokyo, 2017-11-26:
Thanks for explaining.

closed 2017-11-26 by beegee-tokyo