platformio/platform-espressif32 #54: ESP32 toolchain broken??

beegee-tokyo, 2017-11-26:
I run an update today with `platformio update` and if I remember correct it updated the toolchain for ESP32. After the update all builds fail (even the blink example).
Error messages:
```
platformio run
[11/26/17 18:41:32] Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
Collected 26 compatible libraries
Looking for dependencies...
No dependencies
Compiling .pioenvs\esp32dev\FrameworkArduino\Esp.o

Compiling .pioenvs\esp32dev\FrameworkArduino\esp32-hal-gpio.o
Compiling .pioenvs\esp32dev\FrameworkArduino\esp32-hal-i2c.o
Compiling .pioenvs\esp32dev\FrameworkArduino\esp32-hal-ledc.o
Compiling .pioenvs\esp32dev\FrameworkArduino\esp32-hal-matrix.o
Compiling .pioenvs\esp32dev\FrameworkArduino\esp32-hal-misc.o
Compiling .pioenvs\esp32dev\FrameworkArduino\esp32-hal-sigmadelta.o
Compiling .pioenvs\esp32dev\FrameworkArduino\esp32-hal-spi.o
In file included from c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr.h
:148:0,
from c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ext\atomicity.h:35,
from c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\memory:73,
from C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\cores\esp32\Esp.cpp:25:
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:47:9: error: 
'pthread_t' does not name a type
typedef pthread_t __gthread_t;
^
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:48:9: error: 
'pthread_key_t' does not name a type
typedef pthread_key_t __gthread_key_t;
^
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:48:9: error: 
'pthread_key_t' does not name a type
typedef pthread_key_t __gthread_key_t;
^
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:49:9: error: 
'pthread_once_t' does not name a type
typedef pthread_once_t __gthread_once_t;
^
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:50:9: error: 
'pthread_mutex_t' does not name a type
typedef pthread_mutex_t __gthread_mutex_t;
^
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:51:9: error: 
'pthread_mutex_t' does not name a type
typedef pthread_mutex_t __gthread_recursive_mutex_t;
^
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:52:9: error: 
'pthread_cond_t' does not name a type
typedef pthread_cond_t __gthread_cond_t;
^
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:659:19: error
: '__gthread_create' declared as an 'inline' variable
__gthread_create (__gthread_t *__threadid, void *(*__func) (void*),
^

c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:659:19: error
: '__gthread_t' was not declared in this scope
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:659:32: error
: '__threadid' was not declared in this scope
__gthread_create (__gthread_t *__threadid, void *(*__func) (void*),
^
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:659:44: error
: expected primary-expression before 'void'
__gthread_create (__gthread_t *__threadid, void *(*__func) (void*),
^
c:\users\beegee\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:660:5: error:
 expected primary-expression before 'void'
void *__args)
[... 124 lines cut ...]
 
 
Library Manager
===============
Updating Adafruit DHT Unified            @ 1.0.0          [Up-to-date]
Updating Adafruit TSL2561                @ 1.0.0          [Up-to-date]
Updating Adafruit Unified Sensor         @ 1.0.2          [Up-to-date]
Updating ArduinoJson                     @ 5.11.2         [Up-to-date]
Updating DHT sensor library              @ 1.3.0          [Up-to-date]
Updating MQTT                            @ 2.2.1          [Up-to-date]
Updating TFT_eSPI                        @ 0.17.11        [Up-to-date]
Updating Ticker-esp32                    @ 1.0.0          [Up-to-date]
Updating Time                            @ 18ce15b202     [Up-to-date]
Updating WifiManager                     @ 0.12           [Up-to-date]
Updating esp8266FTPServer                @ 97888b3        [Up-to-date]
PS B:\Projects\ESP32\ESP32-Weatherstation - EMQ>  

```
PIO core version:
```
PS B:\Projects\ESP32\ESP32-Weatherstation - EMQ> platformio upgrade
You're up-to-date!
PlatformIO 3.5.0b6 is currently the newest version available.
PS B:\Projects\ESP32\ESP32-Weatherstation - EMQ>  
```

Calzzetta, 2017-11-26:
Same problem here, received a notification to update and now I am getting the same error. The toolchain version was updated 14h ago. I was not able to downgrade the version using the interface from Visual Studio Code. Any idea how to change the toolchain version?

ivankravets, 2017-11-26:
Sorry, fixed! Please `pio update` or `PIO Home > Platforms > Updates`.

closed 2017-11-26 by ivankravets

beegee-tokyo, 2017-11-26:
No need for sorry. You are doing a great job. Do you ever sleep?

Thank you!

Calzzetta, 2017-11-26:
Thank you @ivankravets! I agree with @beegee-tokyo, great job!

ivankravets, 2017-11-26:
Thank you that using PlatformIO. Don't hesitate to report other issues!