platformio/platform-espressif32 #55: Toolchain still broken??? Getting >multiple definition< errors on ArduinoOta example code

beegee-tokyo, 2017-11-28:
Still having problems with PlatformIO and Espressif32.
Because my weatherstation project still fails to compile after the last update of framework and toolchain, I tried to just compile an examples. I chose ArduinoOTA example.
I tried to compile both with **`platform = espressif32`** and with  **`platform = https://github.com/platformio/platform-espressif32.git#feature/stage`**. Both fail with several multiple definition errors.
Also weird is that if I switch between master and staging versions every time I get:
```
PackageManager: Installing toolchain-xtensa32 @ ~2.50200.0
Downloading...
Unpacking...
```
in the compile log. 
And as weird that everytime I run **`platformio update`** I get a line saying
**`Uninstalling toolchain-xtensa32 @ 2.50200.0:    [OK]`**

@ivankravets any idea?
Below additional informations.

**PlatformIO versions**: 
`Home 0.3.2·Core 3.5.0b6`

**Framework, platform and library versions**:
```
PS C:\Users\beegee\Documents\PlatformIO\Projects\171127-155820-arduino-wifiscan> platformio update
Updating tool-scons                      @ 3.20501.2      [Up-to-date]
Updating tool-unity                      @ 1.20302.1      [Up-to-date]
Updating pysite-pioplus                  @ 0.4.2          [Up-to-date]
Updating contrib-piohome                 @ 0.3.2          [Up-to-date]
Updating tool-pioplus                    @ 0.10.14        [Up-to-date]

Platform Manager
================
Platform Espressif 32
--------
Updating espressif32                     @ 0.11.0         [Up-to-date]
Updating tool-esptoolpy                  @ 1.20100.0      [Up-to-date]
Updating framework-arduinoespressif32    @ 1.3.0          [Up-to-date]
Uninstalling toolchain-xtensa32 @ 2.50200.0:    [OK]

Platform Espressif 32 (Stage)
--------
Updating espressif32                     @ 55484a4        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20100.0      [Up-to-date]
Updating framework-arduinoespressif32    @ dac7ae7        [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 1.4.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ 1.20300.1      [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]

Platform Espressif 8266 (Stage)
--------
Updating espressif8266                   @ 08e41d9        [Up-to-date]
Updating framework-arduinoespressif8266  @ 7b09ae5        [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]

Library Manager
===============
Updating Adafruit DHT Unified            @ 1.0.0          [Up-to-date]
Updating Adafruit TSL2561                @ 1.0.0          [Up-to-date]
Updating Adafruit Unified Sensor         @ 1.0.2          [Up-to-date]
Updating ArduinoJson                     @ 5.11.2         [Up-to-date]
Updating DHT sensor library              @ 1.3.0          [Up-to-date]
Updating MQTT                            @ 2.2.1          [Up-to-date]
Updating TFT_eSPI                        @ 0.17.11        [Up-to-date]
Updating Ticker-esp32                    @ 1.0.0          [Up-to-date]
Updating Time                            @ 18ce15b202     [Up-to-date]
Updating WifiManager                     @ 0.12           [Up-to-date]
Updating esp8266FTPServer                @ 97888b3        [Up-to-date]
[... 147 lines cut ...]
ch.o):eh_catch.cc:(.text.__cxa_get_exception_ptr+0x0): first defined here
C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_excepti
on':
cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): multiple definition of `__cxa_end_catch'
c:/users/beegee/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_cat
ch.o):eh_catch.cc:(.text.__cxa_end_catch+0x0): first defined here
C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_excepti
on':
cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): multiple definition of `__cxa_begin_catch'
c:/users/beegee/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_cat
ch.o):eh_catch.cc:(.text.__cxa_begin_catch+0x0): first defined here
C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_excepti
on':
cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): multiple definition of `__cxa_allocate_dependent_exception'
c:/users/beegee/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_all
oc.o):eh_alloc.cc:(.text.__cxa_allocate_dependent_exception+0x0): first defined here
C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_excepti
on':
cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): multiple definition of `__cxa_allocate_exception'
c:/users/beegee/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_all
oc.o):eh_alloc.cc:(.text.__cxa_allocate_exception+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\esp32dev\firmware.elf] Error 1
 [ERROR] Took 85.93 seconds
```

beegee-tokyo, 2017-11-28:
Update:
While typing the issue my platformio core updated to **`Home 0.3.2·Core 3.5.0rc2`**
Error messages still the same.

beegee-tokyo, 2017-11-28:
Another update.
I tried the Arduino Blink example to compile for ESP32.
The original code compiles without errors. But as soon as I add **`#include "ArduinoOTA.h"`** and **`ArduinoOTA.begin();`** into the code it fails with the same **`multiple definition`** errors.
**_Original code:_**
```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```
**_Code that fails to compile:_**
```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"
#include "ArduinoOTA.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
    ArduinoOTA.begin();
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```
Only difference between the two source codes is the addition of **_ArduinoOTA_** code.

Errors during compilation are exactly the same as in my first comment to this issue.

beegee-tokyo, 2017-11-28:
Digging digging digging :) :)

Found in **`platform.py`** of the espressif32 platform
self.packages['toolchain-xtensa32']['version'] = "~2.50200.0"
Changed it into 
self.packages['toolchain-xtensa32']['version'] = "~1.50200.2"
And the code compiles, but throws warning
```
C:\Users\beegee\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\libraries\ArduinoOTA\src\ArduinoOTA
.cpp:1:0: warning: -fstack-protector not supported for this target
#ifndef LWIP_OPEN_SRC
```
But it compiles now without errors and after flashing the code runs.

ivankravets, 2017-11-28:
@me-no-dev do you have any ideas? We updated Arduino Core to the latest version,  in pair with toolchain.

@beegee-tokyo you can use the previous version of espressif32 dev/platform
```ini
[env:esp32dev]
platform = espressif32@0.10.0
board = esp32dev
framework = arduino
```

See all releases https://github.com/platformio/platform-espressif32/releases

me-no-dev, 2017-11-28:
ArduinoOTA yields no such errors when compiled with IDF or ArduinoIDE. Maybe has to do with the order of libs at link time. What is the link command that PIO executes?

ivankravets, 2017-11-28:
@beegee-tokyo please provide full output from `pio run -v` command.

holema, 2017-11-28:
I updated the esspressif32_stage platform to the latest version and sinc them, I get this error:

`Error: Detected unknown package 'espressif32_stage'
`
When I change the platform back to  esspressif32 (v0.11.0) it installs everytime:
`Installing toolchain-xtensa32 @ ~2.50200.0`

And when complling the errormessage is:

`\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o):` In function `__cxx_fatal_exception':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/cxx/./cxx_exception_stubs.cpp:12: multiple definition of 'std::uncaught_exception()'
`
Before I made the update to the latest Version there was no problem with the stage and the normal version of the plattform.

ivankravets, 2017-11-28:
@holema see https://github.com/platformio/platform-espressif32/issues/53

ivankravets, 2017-11-28:
@me-no-dev See attached dump
[dump.txt](https://github.com/platformio/platform-espressif32/files/1510899/dump.txt)

beegee-tokyo, 2017-11-29:
Sorry for late reply (different timezone).
Here are the logs from **`pio run -v`**.
[ok.log](https://github.com/platformio/platform-espressif32/files/1512464/ok.log)  is from compiling **_without_** **`Arduino.h`** and **_without_** **`ArduinoOTA.begin();`**
[bad.log](https://github.com/platformio/platform-espressif32/files/1512465/bad.log) is from compiling **_with_** **`Arduino.h`** and **_with_** **`ArduinoOTA.begin();`**

beegee-tokyo, 2017-11-29:
@ivankravets **`platform = espressif32@0.10.0`** is not working for me because I use SPIFFS which is not available in that version. What would be a working staging version?

holema, 2017-11-29:
Hello,
I changed now the plattform to espressif32. After compiling I get this error:

Linking .pioenvs\nodemcu-32s\firmware.elf
C:\Users\test\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_exception':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/cxx/./cxx_exception_stubs.cpp:12: multiple definition of `std::uncaught_exception()'
c:/users/test/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_catch.o):/builds/id
f/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_catch.cc:138: first defined here
C:\Users\test\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_exception':
cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): multiple definition of `__cxa_call_terminate'
c:/users/test/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_call.o):/builds/idf
/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_call.cc:40: first defined here
C:\Users\test\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_exception':
cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): multiple definition of `__cxa_throw'
c:/users/test/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_throw.o):/builds/id
f/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:62: first defined here
C:\Users\test\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_exception':
cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): multiple definition of `__cxa_rethrow'
c:/users/test/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_throw.o):eh_throw.c
c:(.text.__cxa_rethrow+0x0): first defined here
C:\Users\test\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_exception':
cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): multiple definition of `__cxa_free_dependent_exception'
c:/users/test/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_alloc.o):/builds/id
f/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc:311: first defined here
C:\Users\test\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_exception':
cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): multiple definition of `__cxa_free_exception'
c:/users/test/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_alloc.o):eh_alloc.c
c:(.text.__cxa_free_exception+0x0): first defined here
C:\Users\test\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_exception':
cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): multiple definition of `__cxa_get_exception_ptr'
c:/users/test/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(eh_catch.o):eh_catch.c
c:(.text.__cxa_get_exception_ptr+0x0): first defined here
C:\Users\test\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_exception_stubs.o): In function `__cxx_fatal_exception':

Is there any chance to repaire the framework?

paul-janicaud, 2017-11-29:
@beegee-tokyo I've tried to revert to previous commits with no success.

beegee-tokyo, 2017-11-30:
@paul-janicaud 
As I wrote above I did the following (to be able to continue to work):
Found in **`platform.py`** of the espressif32 platform
**`self.packages['toolchain-xtensa32']['version'] = "~2.50200.0"`**
Changed it into 
**`self.packages['toolchain-xtensa32']['version'] = "~1.50200.2"`**
And the code compiles, but throws warning
```
C:\Users\beegee\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\libraries\ArduinoOTA\src\ArduinoOTA
.cpp:1:0: warning: -fstack-protector not supported for this target
#ifndef LWIP_OPEN_SRC
```
But it compiles now without errors and after flashing the code runs.

paul-janicaud, 2017-11-30:
@beegee-tokyo it seems to work for me for now :) (same warning as you, but my app seems to be running fine)

Thx a lot for sharing this tip,.

holema, 2017-11-30:
@beegee-tokyo THX a lot for the tip.

me-no-dev, 2017-11-30:
I looked through the attached logs. both fail logs look fine at first glance. libs are linked to in proper order, but the ok.log did not include at all the link command so I can not compare. It's rather strange that the toolchain will throw such error

ivankravets, 2017-11-30:
@me-no-dev I see only the 1 difference `-u __cxa_guard_dummy`, I added it to LDFLAGS but that didn't help. I propose temporary to rollback to old toolchain which works very well. 

See my PR https://github.com/espressif/arduino-esp32/pull/894

P.S: Could you give me write permissions to https://github.com/espressif/arduino-esp32 to maintain `platformio-build.py` script? I'll not modify other source files. I have the same permissions for arduino-esp8266 (Ivan gave me). Thanks!

me-no-dev, 2017-11-30:
done and done :)

ivankravets, 2017-11-30:
@me-no-dev Thanks!!! :)  

Do we really need that new toolchain? I have a suspicion that PlatformIO builds libraries and archive them, where Arduino IDE links objects directly. I'll investigate it later.

ivankravets, 2017-11-30:
Fixed in https://github.com/platformio/platform-espressif32/commit/3aa7b8636ff0d6f41adced479468dd118089cdee

closed 2017-11-30 by ivankravets

me-no-dev, 2017-11-30:
you do need the new toolchain for that new feature :) otherwise not really

me-no-dev, 2017-11-30:
@ivankravets ok I might have been given a clue :)
 ```-u __cxa_guard_dummy``` needs to be added (after next rebuild ```-u __cxx_fatal_exception``` also) and you should not put the local libs AFTER the prebuilt IDF ones. Should be before ;)

me-no-dev, 2017-11-30:
wait... i am confused... maybe ```-u __cxx_fatal_exception``` needs to be added now ```-u __cxa_guard_dummy``` should be there

ivankravets, 2017-11-30:
@me-no-dev Wow! People owe you 🍻 

>  you should not put the local libs AFTER the prebuilt IDF ones. Should be before ;)

This worked! https://github.com/espressif/arduino-esp32/commit/d1d56b597af91e1f662d2ca468e24cd9a4199305

me-no-dev, 2017-11-30:
the new toolchain right?

ivankravets, 2017-11-30:
Yes, new toolchain now works. You were right with the order between user libs and Arduino Core (IDF).

So, now `platformio-build.py` is 100% equal to Arduino platforms.txt. Please keep theirs in sync. If something will not work, ping me anytime!

Thanks! 😊

endym, 2018-03-09:
Ping!
Similar problem again. Compiler error: `error: 'pthread_t' does not name a type`

platformio.ini:
```
[env:doorsensor]
platform = espressif32
board = lolin32
framework = espidf, arduino
```
\> pio platform show espressif32
```
espressif32 ~ Espressif 32
==========================
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.

Version: 0.12.0
Home: http://platformio.org/platforms/espressif32
Repository: https://github.com/platformio/platform-espressif32.git
Vendor: https://espressif.com/
License: Apache-2.0
Frameworks: arduino, espidf, pumbaa, simba

Packages
--------

Package toolchain-xtensa32
--------------------------
Type: toolchain
Requirements: ~2.50200.2
Installed: Yes
Version: 2.50200.2
Original version: 5.2.0
Description: GCC for SoC Xtensa L108

Package framework-simba
-----------------------
Type: framework
Requirements: >=12.2.0
Installed: No (optional)

Package framework-arduinoespressif32
------------------------------------
Type: framework
Requirements: ~1.4.0
Installed: Yes
Version: 1.4.0
Original version: None
Description: Arduino Wiring-based Framework (ESP32 Core)

Package framework-pumbaa
------------------------
Type: framework
Requirements: >=2.3.0
Installed: No (optional)

Package tool-espotapy
---------------------
Type: uploader
Requirements: ~1.1.0
Installed: Yes
Version: 1.1.0
Original version: None
Description: This script will push an OTA update to the ESP

Package framework-espidf
------------------------
Type: framework
Requirements: ~2.211.0
Installed: Yes
Version: 2.211.0
Original version: 2.11
Description: Espressif IoT Development Framework with the ESP32

Package tool-esptoolpy
----------------------
Type: uploader
Requirements: ~1.20100.0
Installed: Yes
Version: 1.20100.0
Original version: 2.1.0
Description: ESP8266 and ESP32 serial bootloader utility
```
Switching toolchain-xtensa32 version to 1.50200.2 with @beegee-tokyo's tip helps.

Greetings,
endym

ivankravets, 2018-03-10:
`framework = espidf, arduino` should be one of `espidf` or `arduino`.

endym, 2018-03-10:
Ah, ok. Arduino contains the whole esp-idf API?!
If I select "arduino" as framework, then version 2.50200.x of toolchain-extensa32 is used (defined in `platforms/espressif32/platform.py`). If I select "espidf" then version 1.50200.x is used. That seems to be a problem as well.

Thanks,
endym