platformio/platform-espressif32 #56: Errors installing; platformio platform install espressif32

jorgebnunes, 2017-12-03:
Hi.

I'm facing this issue:

>>>>>>>>>>>>>>------------
PS D:\Visual Studio 2017\Projects\OurSecurity\Blynk> pio update
Updating tool-scons                      @ 3.20501.2      [Up-to-date]
Updating tool-unity                      @ 1.20302.1      [Up-to-date]
Updating pysite-pioplus                  @ 0.4.2          [Up-to-date]
Updating contrib-piohome                 @ 0.4.0          [Up-to-date]
Updating tool-pioplus                    @ 0.11.0         [Up-to-date]

Platform Manager
================
Platform Espressif 8266 (Stage)
--------
Updating espressif8266                   @ 4f59cef        [Up-to-date]
Updating framework-arduinoespressif8266  @ eb49406        [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]

Library Manager
===============
PS D:\Visual Studio 2017\Projects\OurSecurity\Blynk> platformio platform install espressif32
PlatformManager: Installing espressif32
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
espressif32 @ 0.11.0 has been successfully installed!
PackageManager: Installing toolchain-xtensa32 @ ~1.50200.2
Downloading  [------------------------------------]    0%
Please manually remove file `C:\Users\Jorge\.platformio\packages\_tmp_installing-sfpzza-package\toolchain-xtensa32-windows-1.50200.2.tar.gz`
Warning! Package Mirror: [Error 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\Jorge\\.platformio\\packages\\_tmp_installing-sfpzza-package\\toolchain-xtensa32-windows-1.50200.2.tar.gz'
Looking for other mirror...
Downloading  [------------------------------------]    0%
Please manually remove file `C:\Users\Jorge\.platformio\packages\_tmp_installing-tfrkxm-package\toolchain-xtensa32-windows-1.50200.2.tar.gz`
Warning! Package Mirror: [Error 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\Jorge\\.platformio\\packages\\_tmp_installing-tfrkxm-package\\toolchain-xtensa32-windows-1.50200.2.tar.gz'
Looking for other mirror...
Error: Could not install 'toolchain-xtensa32' with version requirements '~1.50200.2' for your system 'windows_amd64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.
PS D:\Visual Studio 2017\Projects\OurSecurity\Blynk>
>>>>>>>>>>>>>>------------

I've tried to uninstall my antivirus ant the result is the same.
Can you help?

Thanks in advance.
Regards,

Jorge

ivankravets, 2017-12-13:
Please contact me@ikravets.com I'll provide further instructions. Thanks!

ivankravets, 2017-12-29:
See http://docs.platformio.org/en/latest/ide/vscode.html#packagemanager-is-unable-to-install-tool

closed 2017-12-29 by ivankravets