platformio/platform-espressif32 #59: partitions.bin file wrong

serioustk, 2017-12-21:
### Hardware:
Board:							        Custom board with ESP32D2WDQ5
Core Installation			Core 3.5.0rc14
IDE name:							Visual Studio Code with Platformio IDE Extension
Flash Frequency:					        80Mhz
Upload Speed:						115200

### Description:
I experience problems flashing an ESP32D2WDQ5 module with Platformio. In difference to all popular boards, this ESP module has only 2 MB Flash. I have generated a custom JSON file for the board:

```json
{
  "build": {
    "core": "esp32",
    "extra_flags": "-DESP32_DEV -DARDUINO_ESP32_DEV",
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dio",
    "ldscript": "esp32_out.ld",
    "mcu": "esp32",
    "variant": "esp32"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "frameworks":  [
    "arduino",
    "espidf"
  ],
  "name": "ESP32 Custom",
  "upload": {
    "flash_size": "2MB",
    "maximum_ram_size": 294912,
    "maximum_size": 655360,
    "require_upload_port": true,
    "resetmethod": "nodemcu",
    "speed": 115200,
    "wait_for_upload_port": true
  },
  "url": "https://en.wikipedia.org/wiki/ESP32",
  "vendor": "Espressif"
}
```

Thus, flashing a simple sketch (see below) results in the following output, the prints are missing:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c
user code done
```

Flashing the same sketch with the Arduino IDE works well and prints as expected on the serial line.

I digged a bit into the problem and found out that the cause of my problem seems to be in the partitions.bin file. This is the flash command that I get from the verbose pio output and that does not work correctly (I broke it up into different lines for better readability):
```
"c:\users\tobia\.platformio\penv\scripts\python.exe"
"C:\Users\tobia\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\esptool.py" 
--before default_reset --after hard_reset --chip esp32 --port "COM3" --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 
0x1000 C:\Users\tobia\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\sdk\bin\bootloader_dio_80m.bin 
0x8000 D:\Dokumente\Repositories\Prototype_0_CoreBoard\.pioenvs\esp32dev\partitions.bin 
0xe000 C:\Users\tobia\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\partitions\boot_app0.bin 
0x10000 .pioenvs\esp32dev\firmware.bin
```

Now, when I replace the partitions file in line 5 by the one created by the Arduino IDE. the sketch works like expected:
```
"c:\users\tobia\.platformio\penv\scripts\python.exe"
"C:\Users\tobia\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\esptool.py" 
--before default_reset --after hard_reset --chip esp32 --port "COM3" --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 
0x1000 C:\Users\tobia\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\sdk\bin\bootloader_dio_80m.bin 
0x8000 C:\Users\tobia\AppData\Local\Temp\arduino_build_862769/sketch_dec20a.ino.partitions.bin  
0xe000 C:\Users\tobia\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\partitions\boot_app0.bin 
0x10000 .pioenvs\esp32dev\firmware.bin
```

Flashing the sketch to a popular NodeMCU ESP-32S (which has got 4MB flash) with platformio works as expedted.

Conclusion: the generation of partitions.bin file in Platformio does not work correctly for ESP32 chips with 2MB (internal) flash.

I don't know where and how the generation of the partitions.bin file is done and would be grateful for any help.

### Sketch:
```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("Tick");
  delay(500);
  Serial.println("Tock");
  delay(500);
}
```

ivankravets, 2017-12-29:
You need `"partitions": "minimal",` in `build` section. See https://github.com/platformio/platform-espressif32/blob/develop/boards/nina_w10.json#L8

serioustk, 2017-12-31:
Hello Ivan, **thank you for your help, that solved my issue!**

I have made some more mistakes that I could solve, but I'd like to share the solution for other guys with the same problem:

```board.json``` file: Simplay follow the docs as http://docs.platformio.org/en/latest/platforms/creating_board.html

When compiling now, one might get an error concerning a missing ```pins_arduino.h``` file. The good but expensive way is to follow this guide: http://docs.platformio.org/en/latest/platforms/creating_platform.html

Or you simply create a folder below ```C:\users\{username}\.platformio\packages\framework-arduinoespressif32\``` that is called like your board and paste a correct ```pins_arduino.h``` file for your board there.

closed 2018-02-09 by ivankravets

tcpipchip, 2018-08-07:
Hi,
How did you solve the Arduino.h missing problem ?

tcpipchip, 2018-08-07:
[image]