platformio/platform-espressif32 #60: PlatformIO can

NRJ-Javor, 2017-12-22:
After updating PlatformIO to Core 3.5.0rc14, it fails to upload firmware to ESP32 Dev Module. Tried uninstalling PlatformIO, Atom, VSCode (yes, I use both IDEs), Python (both ver. 2 and ver. 3) and reinstall VSCode, Atom and PlatformIO. Result is always the same. Project compiles OK. Tried to upload with Arduino - worked.

Here are my specs:
OS: Windows 10 Pro
PlatformIO: Home 0.5.3 Core 3.5.0rc14
Console output:

...
Looking for upload port...
Auto-detected: COM7
Uploading .pioenvs\esp32dev\firmware.bin
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
[--flash_mode {keep,qio,qout,dio,dout}]
[--flash_size FLASH_SIZE]
[--spi-connection SPI_CONNECTION] [--no-progress]
[--verify] [--compress | --no-compress]

[ ...] esptool write_flash: error: argument : Address "Controller" must be a number *** [upload] Error 2 ============================================================================================== [ERROR] Took 3.04 seconds ============================================================================================== The terminal process terminated with exit code: 1
This error happens to both Atom and VSCode, that's why I think it's PlatformIO related.
Even g00gle doesn't know anything regarding "esptool write_flash: error: argument 
: Address "Controller" must be a number"

ivankravets, 2017-12-29:
Please provide `platformio.ini`.

NRJ-Javor, 2017-12-29:
Sure thing, here's the file:

[platformio.zip](https://github.com/platformio/platform-espressif32/files/1593654/platformio.zip)

It's in ZIP to help better uploading. Hope it helps. Everything seems to be OK with it as I last checked. Before the PlatformIO update, everything seemed to be OK. It's just this pair of arguments supplied for the function. I found the source of the exception, it's in "esptool.py", line 2390, where the actual exception is thrown, due to faulty pair of arguments supplied.

ivankravets, 2017-12-29:
Please open VSCode PIO IDE Terminal and type `pio run -v -t upload` and paste output on https://hastebin.com and share link here.

NRJ-Javor, 2018-01-02:
Hey :)
Sorry for the late reply, but was out for holidays. I pasted the output as you advised me and here's the url: [https://hastebin.com/utemesohar.tex](url).
Hope it helps. Please, I need this fixed in order to upload firmware to my controller :(
Thanks in advance.

ivankravets, 2018-01-06:
Is "USER" a real username? Does it contain spaces or non ascii-chars?

NRJ-Javor, 2018-01-08:
No, USER is a replacement for my user on Windows. It is one word, no spaces, written in English, no non-standard letters.
Btw I found a solution to my problem and posted the solution here: [https://community.platformio.org/t/uploading-to-esp32-failed-solved/3181](url)
Seems it's not a PlatformIO feature but Espressif32 Platform (both just updated at the same time)...
For anyone who's interested in solving the problem, check the link above.

cyberman54, 2018-01-21:
I have exactly the same problem, since a few weeks.
And yes, my Windows10 user name contains a "blank", and i can't change this :-(
Probably this causes wrong parsing of arguments when esptool.py i called?

ivankravets, 2018-01-23:
Could you try a staging version of dev/platform? Does it work now?

- http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version

closed 2018-02-09 by ivankravets