platformio/platform-espressif32 #62: Support for uploading SPIFFS files to ESP32 boards?

huming2207, 2018-01-05:
Hi all,

I've submitted this issue some days ago to the main repo and I was thinking this is a bug while @ivankravets said it's not. It is not yet implemented. (See: https://github.com/platformio/platformio-core/issues/1269#issuecomment-355081485)

So...could you guys implement this upload function these days? It is very useful for some of the ESP32 learners like me.  Thanks by the way.

Regards,
Jackson

baggior, 2018-01-27:
I'm very interested too
beacuse actually I'm forced to work in the arduino IDE instead just for uploading SPIFFS on ESP32

ivankravets, 2018-01-27:
@baggior do you mean that it works in Arduino IDE for ESP32?

huming2207, 2018-01-28:
@ivankravets Yea of course, currently I created a pseudo *.ino file and open that directory with Arduino IDE. Then it can upload the SPIFFS for me. But the problem is that I have to open another Java program instead of using CLion and/or VSCode + PlatformIO (and literally, Arduino IDE sucks, no advanced linting, no full Asian chars support even in the year of 2018, while it uses 200+ MB more RAM and 400+ MB storage but just doing a notepad job lol). 

I would like to implement this function later this week if I have enough free time. I'll open a PR later if I can.

baggior, 2018-01-28:
@ivankravets yes, there is a small java class that builds spiffs and uploads for esp32
here is the link [https://github.com/me-no-dev/arduino-esp32fs-plugin]( https://github.com/me-no-dev/arduino-esp32fs-plugin)

I've reverseenginered the code and I make a batch file for windows that does the same for my ESP32DEV and it works

but I dont know how to integrate it into platformio as it actually already works for the ESP8266 platforms

here is my batch
[upload_spiffs.cmd.txt](https://github.com/platformio/platform-espressif32/files/1670987/upload_spiffs.cmd.txt)

sticilface, 2018-04-12:
would like to have this too.

JoarGjersund, 2018-04-26:
any updates on this?

ivankravets, 2018-05-16:
Please reopen if you still need help

huming2207, 2018-05-16:
Hmmmm sorry I'm busy dealing with my school assignment these days and I'll try it later in June.

sarfata, 2018-05-16:
@ivankravets does not seem to work for me:

`platformio.ini`:
```
[platformio]
data_dir = webfiles

[env:esp32]
src_filter = +<common/*>,+<esp32/*>
build_flags =
    ${common.build_flags}
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
framework = arduino
board = esp32dev
```

```
$ platformio run -vv -e esp32 -t uploadfs
"mkspiffs_espressif32_arduino" -c webfiles -p 256 -b 4096 -s 1503232 .pioenvs/esp32/spiffs.bin
/bundle.js.gz
/index.html
/style/style.css
MethodWrapper(["uploadfs"], [".pioenvs/esp32/spiffs.bin"])
Auto-detected: /dev/cu.SLAB_USBtoUART
"/usr/bin/python" "/Users/thomas/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32 --port "/dev/cu.SLAB_USBtoUART" --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_size detect 2691072 0x10000 .pioenvs/esp32/spiffs.bin
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
[--flash_mode {keep,qio,qout,dio,dout}]
[--flash_size FLASH_SIZE]
[--spi-connection SPI_CONNECTION] [--no-progress]
[--verify] [--compress | --no-compress]
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '0x10000'
*** [uploadfs] Error 2
```

If I remove `0x10000` it works:

```
"/usr/bin/python" "/Users/thomas/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32 --port "/dev/cu.SLAB_USBtoUART" --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_size detect 2691072 .pioenvs/esp32/spiffs.bin
esptool.py v2.3.1
Connecting........_____..
Chip is ESP32D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1503232 bytes to 245542...
Wrote 1503232 bytes (245542 compressed) at 0x00291000 in 21.7 seconds (effective 554.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

I think the problem is here: https://github.com/platformio/platform-espressif32/blob/5be3821691491ee80a7ab8a79c98bea603767a15/builder/main.py#L239

You need to remove `0x10000` in the esp32 version of that file.

ivankravets, 2018-05-18:
@valeros could you help?

ivankravets, 2018-05-25:
See docs http://docs.platformio.org/en/latest/platforms/espressif32.html#uploading-files-to-file-system-spiffs

nelfata, 2018-05-27:
Hi, I am trying to use the latest PIO with ESP32 and SPIFFS. Following the instructions above I don't seem to have it working (see output below)

OUTUPUT:
```
platformio run -t buildfs
[05/27/18 14:30:05] Processing esp32stage (framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/stage; board: esp32dev)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 (Stage) > Espressif ESP32 Dev Module
SYSTEM: ESP32 160MHz 288KB RAM (1.25MB Flash)
Converting FS.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Log>
|   |-- <common>
|-- <common>
|-- <FS> v1.0
|-- <SPIFFS> v1.0
|   |-- <FS> v1.0
*** Do not know how to make File target `buildfs' (C:\Users\NEF\Desktop\NextCloud\Clones\LogicalCircuits\EverSound\Firmware\arduino\test\ESP32\FS\buildfs).  Stop.
```

ivankravets, 2018-05-27:
Could you try `pio update`?

nelfata, 2018-05-27:
I did but still it does not work.

nelfata, 2018-05-27:
Here is my platformio.ini:

```
[platformio]
env_default = esp32stage
src_dir     = ./
lib_dir     = ../../../common/libraries
data_dir    = ./data

[env:esp32stage]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage

board     = esp32dev
framework = arduino

; set frequency to 160MHz
board_build.f_cpu = 160000000L
; set frequency to 80MHz
board_build.f_flash = 80000000L
;board_flash_mode = qio
upload_speed = 921600

```

nelfata, 2018-05-27:
Version:
`PlatformIO, version 3.5.3rc1`

nelfata, 2018-05-29:
Do you have any timeline on resolving this issue?

nelfata, 2018-05-29:
I am using Windows7, and when I removed `C:\Users\NEF\.platformio\platforms` that worked.
I did not complete the testing but at least the command: `platformio run -t buildfs` works with no issues.

Thanks for the follow up.

closed 2018-05-30 by ivankravets

grownseed, 2018-06-20:
`Detected non-PlatformIO `data_dir` option` on `PlatformIO, version 3.6.0a2` when specifying `data_dir` for my own env. If I put it under `[platformio]` it works. Is this the only place where it can be defined at the moment?

I'm also wondering if there is a way of making the SPIFFS upload a part of the build process, instead of having to run `pio run --target uploadfs` independently?

Thanks!

ivankravets, 2018-06-20:
Targets

- `buildfs`
- `uploadfs`

grownseed, 2018-06-21:
@ivankravets thanks! Any thoughts on having `data_dir` defined in user-defined configs, as opposed to having to have it in `[platformio]`? Or perhaps I'm misunderstanding how this is supposed to work?

grownseed, 2018-06-21:
Great, thank you, and my apologies if these are stupid questions, I'm very new to ESP dev and PIO, and still finding my bearings.

zhivko, 2018-07-02:
Althought this is closed it belongs here:

I am getting:
```

C:\Users\klemen\Dropbox\Voga\BleVogaLifter-esp32-DRV8703Q>pio --version
PlatformIO, version 3.5.3

[07/02/18 14:40:35] Processing esp32 (platform: https://github.com/platformio/platform-espressif32.git#feature/stage; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Detected non-PlatformIO `// platform` option in `[env:esp32]` section
Detected non-PlatformIO `// build_flags` option in `[env:esp32]` section
Detected non-PlatformIO `// upload_port` option in `[env:esp32]` section
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 (Stage) > Espressif ESP32 Dev Module
SYSTEM: ESP32 160MHz 288KB RAM (1.25MB Flash)
DEBUG: CURRENT(minimodule) EXTERNAL(minimodule, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h)
Converting Server.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> v1.1.1 #44d95ae
|   |-- <AsyncTCP> v1.0.1 #f34763c
|   |-- <FS> v1.0
|   |-- <WiFi> v1.0
|-- <FS> v1.0
|-- <AsyncTCP> v1.0.1 #f34763c
|-- <WiFi> v1.0
|-- <SPI> v1.0
|-- <SPIFFS> v1.0
|   |-- <FS> v1.0
|-- <ArduinoOTA> v1.0
|   |-- <ESPmDNS> v1.0
|   |   |-- <WiFi> v1.0
|   |-- <WiFi> v1.0
|   |-- <Update> v1.0
Building SPIFFS image from 'data' directory to .pioenvs\esp32\spiffs.bin
/index.html
/rose.svg
/script.js
Looking for upload port...
Use manually specified: 192.168.43.146
Uploading .pioenvs\esp32\spiffs.bin
esptool.py v2.3.1
Traceback (most recent call last):
File "C:\Users\klemen\.platformio\packages\tool-esptoolpy\esptool.py", line 2637, in <module>
_main()
File "C:\Users\klemen\.platformio\packages\tool-esptoolpy\esptool.py", line 2630, in _main
main()
File "C:\Users\klemen\.platformio\packages\tool-esptoolpy\esptool.py", line 2355, in main
esp = chip_class(args.port, initial_baud, args.trace)
File "C:\Users\klemen\.platformio\packages\tool-esptoolpy\esptool.py", line 193, in __init__
self._port = serial.serial_for_url(port)
File "c:\python27\lib\site-packages\serial\__init__.py", line 88, in serial_for_url
instance.open()
File "c:\python27\lib\site-packages\serial\serialwin32.py", line 62, in open
raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '192.168.43.146': WindowsError(2, 'The system cannot find the file specified.')
*** [uploadfs] Error 1
=========================================================================== [ERROR] Took 1.75 seconds ===========================================
```

ivankravets, 2018-07-02:
@zhivko fixed in https://github.com/platformio/platform-espressif32/commit/b3bcc30858164ee70b40c08f88666842e9593c81

The final release is coming today.

zhivko, 2018-07-02:
I gues I could do ```pio update``` to get update?

DarvosBR, 2019-04-03:
> @ivankravets yes, there is a small java class that builds spiffs and uploads for esp32
> here is the link https://github.com/me-no-dev/arduino-esp32fs-plugin
>
> I've reverseenginered the code and I make [...]

@baggior I just wanna let you know! Thank you!