platformio/platform-espressif32 #63: Update to the new stable version 3.0 of esp-idf

rmh78, 2018-01-09:
https://github.com/espressif/esp-idf/tree/release/v3.0

ivankravets, 2018-01-10:
We support only stable software. Please reopen this issue when ESP-IDF 3.0 will be released.

steveatinfincia, 2018-04-29:
As of a few days ago, ESP-IDF 3.0 is released as stable. They had a `release/v3.0` branch for a long time, but the `v3.0` tag is tied to the actual public release. 

https://github.com/espressif/esp-idf/releases/tag/v3.0

ivankravets, 2018-05-05:
Resolved in https://github.com/platformio/platform-espressif32/commit/218faa1b7363e3e9c26ff0d259a2d6733810d468

ivankravets, 2018-05-05:
Please re-test with the latest upstream version http://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions

FoxKeys, 2018-05-07:
Tried to build upstream. Got:

In file included from D:/users/ad/.platformio/packages/framework-espidf/components/mbedtls/port/include/mbedtls/esp_config.h:2665:0,
from D:/users/ad/.platformio/packages/framework-espidf/components/mbedtls/include/mbedtls/sha256.h:29,
from D:\users\ad\.platformio\packages\framework-espidf\components\bootloader_support\src\bootloader_sha.c:22:
D:/users/ad/.platformio/packages/framework-espidf/components/mbedtls/include/mbedtls/check_config.h:73:2: error: #error "MBEDTLS_CTR_DRBG_C defined, but not all prerequisites"
#error "MBEDTLS_CTR_DRBG_C defined, but not all prerequisites"

How should i fix this?

ivankravets, 2018-05-07:
@FoxKeys could you provide a simple code to reproduce this issue?

FoxKeys, 2018-05-07:
Just empty project...
extern "C" void app_main(void)
{
}

Maybe, problem in configs (but, same works fine with platform = espressif32)

Linked project.
[ESP32BLENew.zip](https://github.com/platformio/platform-espressif32/files/1980171/ESP32BLENew.zip)

ivankravets, 2018-05-07:
@FoxKeys could you try new config? https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-http-request/src/sdkconfig.h

hammondeggs, 2018-05-07:
Also ran into these same issues - have changed to use the new sdkconfig.h file and can confirm the software now at least builds. Thank you...!

ivankravets, 2018-05-07:
@valeros should we automatically update user's `sdkconfig.h`? Or, maybe some warning?

FoxKeys, 2018-05-07:
With your config - build success.
But. Configs are very, very different. And, how to adopt mine?
Need some sort of "upgrade" i suppose... At least, i can't convert old to new "by hand" (not with my knowledge)...

ivankravets, 2018-05-07:
Did you apply any changes to the previous config?

FoxKeys, 2018-05-07:
Just things i have found:
At least, this options required
#define CONFIG_MBEDTLS_AES_C 1
#define CONFIG_BLUEDROID_PINNED_TO_CORE 0
#define CONFIG_TCP_OVERSIZE_MSS 1
#define CONFIG_TIMER_TASK_STACK_SIZE 3584
#define CONFIG_WL_SECTOR_SIZE 4096

But, this is not enough. Build failed on 
Compiling .pioenvs\upstream\fatfs\src\ff.o
D:\users\ad\.platformio\packages\framework-espidf\components\fatfs\src\ff.c:543:37: error: 'TBL_CT1' undeclared here (not in a function)
static const BYTE ExCvt[] = MKCVTBL(TBL_CT, FF_CODE_PAGE);
^
D:\users\ad\.platformio\packages\framework-espidf\components\fatfs\src\ff.c:407:26: note: in definition of macro 'MERGE_2STR'
#define MERGE_2STR(a, b) a ## b
^
D:\users\ad\.platformio\packages\framework-espidf\components\fatfs\src\ff.c:543:29: note: in expansion of macro 'MKCVTBL'
static const BYTE ExCvt[] = MKCVTBL(TBL_CT, FF_CODE_PAGE);
^
*** [.pioenvs\upstream\fatfs\src\ff.o] Error 1

And, here i don't know what is next required...

FoxKeys, 2018-05-07:
> Did you apply any changes to the previous config?
I'm user your (totally replaced mine). Build success. But, i'm not sure it will _work_ (because can be different chip settings, for example)
Compare is too complex for me (more than 100 lines different, and, most is "WTF??" (not documented))

FoxKeys, 2018-05-07:
So, need some sort of migration tool (or, at least, manual?)

FoxKeys, 2018-05-08:
One more thing. Not sure it is related to this topic, but maybe?
Add
build_flags = -fexceptions
into platformio.ini

Build successfull. But, doesn't works as expected (infinite reboot loop).

Test code
```
try {
        ESP_LOGI( GATTC_TAG, "%s - try - before throw", __func__ );
        throw 1;
        ESP_LOGI( GATTC_TAG, "%s - try - after throw", __func__ );
    } catch (...) {
        ESP_LOGI( GATTC_TAG, "%s - catch", __func__ );
    }
```

versamodule, 2018-05-09:
@ivankravets , when you said "@FoxKeys could you try new config? https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-http-request/src/sdkconfig.h" 

I tried this and it then compiled. But my question is why when creating a new IDF project didn't it do this already ?

FoxKeys, 2018-05-09:
> @ivankravets , when you said "@FoxKeys could you try new config? https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-http-request/src/sdkconfig.h"

Yes.
But, exceptions doesn't works under new config too (although the code compiles successfully).
ESP-IDF anounced exceptions support

> C++ exception support (configurable), including support in libstdc++ (#681, plus follow-up work)

Maybe, i misunderstood something?

ivankravets, 2018-05-09:
@versamodule fixed by @valeros in https://github.com/platformio/platform-espressif32/commit/1ca3930a8f080fece3257b5e29c9d51dec532c14

ivankravets, 2018-05-24:
Related issue for `exceptions`  https://github.com/platformio/platform-espressif32/issues/76

closed 2018-05-24 by ivankravets