platformio/platform-espressif32 #66: esp_bt.h dependency

ozanoner, 2018-01-23:
Hi,
I have the following error:
Compiling .pioenvs\esp32doit-devkit-v1\bootloaderSupport\src\bootloader_flash.o
src\gattc_demo.c:28:20: fatal error: esp_bt.h: No such file or directory

platform: espressif32
framework: esp-idf
original: https://github.com/espressif/esp-idf.git

pls see the attached file for details.
Regards..
[error.txt](https://github.com/platformio/platform-espressif32/files/1657328/error.txt)

blackketter, 2018-02-26:
Same problem here.  Any suggestions @ivankravets ?

Thanks!

ToeffToeff, 2018-02-26:
The issue is caused by different esp-idf versions PIO is using and your code is written for.

Platformio currently uses esp-idf v2.1 where the Bluetooth header file is named bt.h and your code assumes esp_bt.h
The rename was done in v3.0 of esp-idf as you can see at the second last point of "Breaking Changes" for v3.0 here:
https://github.com/espressif/esp-idf/releases

Depending on the features your project uses a simple renaming of the header file in your code to bt.h may fix the issue.
But if you use some new features of esp-idf v3.0 the only way I figured out was to install esp-idf v3.0 standalone and compile your program with that.

There is already a thread in the PIO Community forum about esp-idf v3.0 in PIO (and some not working workaround attempts) so it looks like we have to wait for the PIO team to bring up v3.0 of esp-idf officially.
FYI the thread is this one: https://community.platformio.org/t/using-esp32-idf-master-branch/2786

blackketter, 2018-02-26:
thanks @ToeffToeff ! I made a esp_bt.h file to get the sample code compiling.  Looking forward to support for v3.0

wegunterjr, 2018-05-01:
@blackketter where do you create the esp_bt.h?

blackketter, 2018-05-01:
I believe I just created a esp_bt.h file and put it in my src directory.  All it contained was one line that was:

#Include “bt.h”

wegunterjr, 2018-05-01:
alright, tried that but still getting this error:  
`.platformio\lib\ESP32 BLE Arduino_ID1841\src\BLEDevice.cpp:14:44: fatal error: esp_bt.h: No such file or directory`

wegunterjr, 2018-05-01:
[image]

ivankravets, 2018-05-05:
Does it work now with the latest upstream version? 
* http://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions

wegunterjr, 2018-05-05:
It compiles and I can upload to the huzzah32...and it runs!!

ivankravets, 2018-05-05:
Thank you so much for the report! The final release of dev/platform is planned next week. 

Stay tuned at https://twitter.com/PlatformIO_Org

closed 2018-05-05 by ivankravets