platformio/platform-espressif32 #69: SPIFFS read does not work

9a4gl, 2018-02-10:
Looks like after last update of espressif 32 platform in PlatformiIO there is problem reading SPIFFS. 

I found this https://github.com/espressif/arduino-esp32/issues/1068

ivankravets, 2018-02-10:
Could you open `~/.platformio/packages/framework-arduinoespressif32/tools/platformio-build.py` and  replace `env.Prepend(LIBS=libs)` with `env.Append(LIBS=libs)` in 178 line. Then clean project and re-flash . Does it work now?

9a4gl, 2018-02-10:
No, it does not work. 
But it works with platform = espressif32@0.11.1

ivankravets, 2018-02-10:
Let's wait when issue will be fixed in https://github.com/espressif/arduino-esp32/issues/1068

/cc @me-no-dev

9a4gl, 2018-02-10:
I do not know if the issue is exactly the same. What I observed is that SPIFFS mounts, it can list files. But reading file is problem. In log I see:
1. json parser failed to parse from File
2. fsbrowser list files, but does provide empty content of file.

ivankravets, 2018-02-10:
Please install Arduino IDE https://github.com/espressif/arduino-esp32#installation-instructions and use the latest source code.

Does it work with Arduino IDE?

9a4gl, 2018-02-10:
Arduino latest - Wrote 789632 bytes (447258 compressed)  - does not work
Platformio latest (0.12.0)  - Compressed 731248 bytes to 435798 - does not work
Platformio 0.11.1 - Compressed 635248 bytes to 401350 - works
I also do not like much that my app increased by 100k

ivankravets, 2018-02-10:
Thanks! Let's wait when this bug will be fixed in Arduino Core. Please ping me and I'll update "release" package.

Nevertheless, you can stay with previous version as you mentioned before `platform = espressif32@0.11.1`

baggior, 2018-02-11:
I think the problem is related to new spiff configuration from IDF

https://github.com/espressif/arduino-esp32/issues/1022

lonerzzz, 2018-02-12:
As per (https://github.com/espressif/esp-idf/issues/1610#issuecomment-365022594), @igrr has provided a 0.2.2 version () of mkspiffs that can be downloaded on its own (https://github.com/igrr/mkspiffs/releases/tag/0.2.2) until @me-no-dev spins a new version.

ivankravets, 2018-05-09:
We upgraded mkspiffs to 0.2.3. Please re-test with the latest upstream version http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version

closed 2018-05-09 by ivankravets

GiampaMan, 2020-07-21:
Hi!

Maybe this issue is old, but I have this exact same issue but this is happening when I pass in PlatformIO from version 1.9.0 to 1.12.4 on ESP32 platform. The only way I realized this was happening was when I saw errors when parsing on ArduinoJson. Then I just tried to print directly the SPIFFS file on serial port but it was not possible. Can anyone knows how to do in this case?

I will appreciate any help, I'm a beginner.

Thanks