platformio/platform-espressif32 #70: custom framework: updated espidf

boneskull, 2018-02-11:
Hi, sorry @ivankravets, I think I've asked this type of question before, but I can't find the answer.

I understand how to create a custom *platform* and a custom *board*.  But what I'd like to do is use a newer version of `espidf`.  I'm assuming I'll need a modified `platform-espressif32` to do so.

I'm looking at `~/.platformio/packages/framework-espidf`, which *appears* to be a fork of [espressif/esp-idf](/espressif/esp-idf), with a `package.json` added.  I can't find any related repository for it in the [platformio org](/platformio).  So I'm at a loss.

My use case is building [a project](/nkolban/duktape-esp32) using ESP-IDF's new-ish SPIFFS support.

ToeffToeff, 2018-02-12:
Hi, I'm also really interested in this!

I have to use the Bluetooth SPP API in my Project which isn't available in `espidf` version 2.211.0 (currently the actual one in PIO) - only in 3.0.

I tried some workarounds with copying the last version of `espidf` files to `packages` folder like described [here](https://community.platformio.org/t/platformio-is-not-updating-esp-idf-files/2733) but ran into the same issue thats described there and found no solution for them.

I installed `esp-idf 3.0` + toolchain manually (w/o PIO) and it works, but PIO is so much more comfortable and i would love to use it also for this project! :)

ivankravets, 2018-05-05:
You can take a look at what is @valeros did for ESP-IDF 3.0 https://github.com/platformio/platform-espressif32/commit/218faa1b7363e3e9c26ff0d259a2d6733810d468

The same he will do with SPIFFS. Please take a look at the latest commits https://github.com/platformio/platform-espressif32

closed 2018-05-05 by ivankravets