platformio/platform-espressif32 #71: Select task to run

haberturdeur, 2018-02-16:
Hi,
when I'm using PIO on VSCode on windows, when I try to build/upload It shows me "Select task to run" with three options: 
[image]

(and the build/upload won't start)

ivankravets, 2018-05-05:
Please check your `platformio.ini`. Does it work if you type in VSCode terminal `pio run`?

closed 2018-05-05 by ivankravets

rob-lutrons, 2019-08-09:
I had this too. After comparing with a working project I noticed that the com port wasnt set
[image]

Clicking on this opens a drop down at the top of the editor. Selecting a port solved my problem.

ivankravets, 2019-08-09:
Please uninstall conflicted Arduino extension.