platformio/platform-espressif32 #72: PLatformIO arduino-esp32 compiling wrong library

Brunez3BD, 2018-02-23:
This question is moved from [https://github.com/espressif/arduino-esp32/issues/1121](url)

Hi,
I'm trying the WiFi example `arduino-esp32/libraries/WiFi/examples/WiFiScan/WiFiScan.ino` with PlatformIO, but when compiling it raise the error:
`'class WiFiClass' has no member named 'mode' `
How can I solve this?

lookinano, 2018-05-11:
I can confirm the issue. It seems as if there is no library with WiFi.h Header file available for espressif32 platform online: https://platformio.org/lib/search?query=header:WiFi.h
So that the lookup in the Library Manager fails.

For now I put the library folder from platform-package (_.platformio/packages/framework-arduinoespressif32/libraries/WiFi_) into my _PROJECT_PATH/lib_ folder as a workaround.

ivankravets, 2018-05-24:
Please re-test with the latest upstream version http://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions

closed 2018-05-24 by ivankravets

danielkucera, 2018-11-10:
I had to uninstall all libraries which have WiFi.h to force PIO to use the one from arduino-esp32. Is there any way how to force it?