platformio/platform-espressif32 #76: Bluetooth compilation issue with Platformio

ayhoung, 2018-03-17:
I'm trying to include the below libraries with the below code in Platform.io
```
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

void setup() {

}

void loop() {

}
```

This works perfectly in the Arduino IDE, but for some reason I'm getting the below error when compiling with Platformio:
```
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 (Stage) > Espressif ESP32 Dev Module
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 21 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 BLE Arduino> v0.4.9
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEClient.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLERemoteCharacteristic.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLERemoteDescriptor.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLERemoteService.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEScan.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLESecurity.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEServer.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEService.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEServiceMap.cpp.o
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp: In member function 'BLERemoteService* BLEClient::getService(BLEUUID)':
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp:336:12: error: exception handling disabled, use -fexceptions to enable
throw new BLEUuidNotFoundException;
^
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLERemoteDescriptor.cpp: In member function 'std::__cxx11::string BLERemoteDescriptor::readValue()':
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLERemoteDescriptor.cpp:64:34: error: exception handling disabled, use -fexceptions to enable
throw BLEDisconnectedException();
^
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLERemoteCharacteristic.cpp: In member function 'std::__cxx11::string BLERemoteCharacteristic::readValue()':
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLERemoteCharacteristic.cpp:439:34: error: exception handling disabled, use -fexceptions to enable
throw BLEDisconnectedException();
^
*** [.pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLERemoteDescriptor.cpp.o] Error 1
*** [.pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEClient.cpp.o] Error 1
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLERemoteService.cpp: In member function 'BLERemoteCharacteristic* BLERemoteService::getCharacteristic(BLEUUID)':
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLERemoteService.cpp:156:37: error: exception handling disabled, use -fexceptions to enable
throw new BLEUuidNotFoundException();
^
*** [.pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLERemoteCharacteristic.cpp.o] Error 1
*** [.pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLERemoteService.cpp.o] Error 1
```

I've used both configs
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp32dev
framework = arduino
```
and
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
```

but both don't seem to work. Has anyone else run into this issue?

wegunterjr, 2018-04-30:
I am having this same problem. I converted my code that compiles in Arduino IDE manually to a cpp file in PlatformIO, but it will not compile. I added the `-x gcc`  as indicated in this following step: `Open .gcc-flags.json file from the Initialized/Imported project and add -x c++ flag at the beginning of the value of gccDefaultCppFlags field:`

wegunterjr, 2018-05-04:
here is what I did to get it to build:

> [env:featheresp32]
> //platform =espressif32@0.11.1
> platform = https://github.com/platformio/platform-espressif32.git#feature/stage
> lib_extra_dirs = ~/Documents/Arduino/libraries
> board = featheresp32
> framework = arduino
> 

Here is my folder structure.  I also had to add a file to src `esp_bt.h` to reference `bt.h`.
[image]

garvamel, 2018-05-10:
I fixed it by replacing `packages/framework-arduinoesspresif32` contents with the same ones downloaded from [](https://github.com/espressif/arduino-esp32) and adding `build_flags = -fexceptions` to `platformio.ini`. Everything builds fine now

ivankravets, 2018-05-25:
Works with the latest upstream version http://docs.platformio.org/en/latest/platforms/espressif32.html#upstream

closed 2018-05-25 by ivankravets

FoxKeys, 2018-05-25:
> @FoxKeys Does this code work with Arduino IDE?
This is not arduino code. This is for ESP-IDF framework.
And, no, it doesn't work in latest too (no updates awailable, but how to check current version?)

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = espidf
build_flags = -fexceptions
lib_ldf_mode = chain+
lib_compat_mode = off

FoxKeys, 2018-05-25:
P.S. Installed version 9080ace

ivankravets, 2018-05-25:
@FoxKeys please file a separate issue https://github.com/platformio/platform-espressif32/issues with all details and code example.

FoxKeys, 2018-05-25:
Created [https://github.com/platformio/platform-espressif32/issues/88](url)