platformio/platform-espressif32 #78: ESP32 does not accept OTA

williamesp2015, 2018-03-26:
Platformio.ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_port = 192.168.1.6

13:50:30 [INFO]: Upload size: 599168
Sending invitation to 192.168.1.6
13:50:30 [INFO]: Waiting for device...
13:50:40 [ERROR]: No response from device
*** [upload] Error 1

DJManas, 2018-04-16:
**What port are you using? Default or are you overriding it with ArduinoOTA.setPort?**

I have example sketch copied from BasicOTA.ino put into method, which I call in setup after calling wifi connect:

```
void prepareOTA() {
  // Port defaults to 3232
  ArduinoOTA.setPort(3232);
  // Hostname defaults to esp3232-[MAC]               // ArduinoOTA.setHostname("myesp32");
  // No authentication by default                     // ArduinoOTA.setPassword("admin");
  // Password can be set with it's md5 value as well  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");
  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";
    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });

  ArduinoOTA.onEnd([]() { Serial.println("\nEnd"); });

  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });

  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR)         Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR)   Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR)     Serial.println("End Failed");
  });

  ArduinoOTA.begin();
}
```

Working pretty fine, but I have to have configuration like this:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_port = 192.168.100.41
upload_flags = --port=3232
```

Of course you have to have ArduinoOTA.handle() in your loop. I have this to be enabled 30 seconds after boot, that is the otaFlag is being disabled.
```
  // Handle OTA you have 30 seconds
  while (otaFlag) {
    ArduinoOTA.handle();

    // Wait 60 seconds after start to OTA...
    if ((millis() / 1000) > 30)
      otaFlag = false;
  }
```

**Do you have clear network path to the ESP?**
I mean the problem might be, that the initial connection is made from your PC, but then it establishes server on your PC, which might be blocked by firewall on your pc.

ivankravets, 2018-05-05:
Thanks! I updated docs at http://docs.platformio.org/en/latest/platforms/espressif32.html#over-the-air-ota-update

closed 2018-05-05 by ivankravets