platformio/platform-espressif32 #79: Upload error for the ESP-IDF

williamesp2015, 2018-03-27:
My PlatformIO is updated ATOM is 1.2.5.0 x64
I had no problem with Esp32 using Arduino but when I tried to use PlatformIO examples: espidf-hello-world I could build without error and when tried to upload I got this error even with stage version

esptool write_flash: error: argument

: [Errno 2] No such file or directory: 'C:\~Platform
Io\~MYESP32\~Final'
My platofrmIo.ini is:
[env:esp32dev]
platform = espressif32
;platform = https://github.com/platformio/platform-espressif32.git#feature/stage
framework = espidf
board = esp32dev

Gouthamm4G, 2018-04-18:
I ran into a similar issue, maybe and I solved it just now. Does your project path (the complete path) have spaces in them? The build script seems to consider it to be a path without spaces. I have a fix for it, but I don't think you seem to be having the same problem. What should be the expected directory there for you? Its the path of either the bootloader or the firmware, but you might as well consider it as the path of your platformio.ini file.
<br><br>
For me, the error was so,
```
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\Gouthamm'
*** [upload] Error 2

```

jafr0822, 2018-05-18:
I'm using the feature/stage branch and I'm still having the same error:
"esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory:"
My project directory does have spaces in it, so, I think that it's the same issue.

ivankravets, 2018-05-24:
Please re-test with the latest upstream version http://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions

closed 2018-05-24 by ivankravets

Staars, 2018-10-07:
This is another incarnation of "path name containing space leads to an error". I have an ESP32 and wanted to use my iCloudDrive.

The path to an iCloudDrive begins with:
'/Users/<name>/Library/Mobile Documents/.....
So there is a space between Mobile and Documents.

Opening a project and building is not an issue, but uploading leads to:
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/Users/<name>/Library/Mobile'
*** [upload] Error 2

I already created a symlink and use it to open the project, but the error is identical and the original path name shows up.
I can recreate the error with other folders with spaces in the path (i.e. "New folder").

Version:
Home 1.0.2·Core 3.6.1a5

Atom and VS-Code do not make a difference.

I would call this a minor problem as I can use Dropbox for the same purpose, but it is still not solved.

Where is the last python script in the chain located that starts the esptool? Maybe we can fix it there.

commit 5be38216 (2018-05-16): Escape a path for flash extra images // Resolve #79
commit 2cc5bd91 (2018-05-16): Merge branch 'develop' into feature/stage

* develop:
  Fix typo
  Escape a path for flash extra images // Resolve #79