platformio/platform-espressif32 #80: ESP32 example OTA sketch

DJManas, 2018-04-16:
Hello,

dunno if this on PlatformIO or on ESP-IDF framework, but when I create new project, take [this example](https://github.com/espressif/esp-idf/blob/master/examples/system/ota/main/ota_example_main.c) it fails to compile. I had found out, that the esp-idf is newer than in PlatformIO, so I took from .platformio/packages/framework-idf... it compiled, but today when I try to compile it for ESP32-DEV or WEMOSBAT it fails to compile with this error:

```
.pioenvs/esp32dev/libapp_update.a(esp_ota_ops.o):(.literal.esp_ota_end+0x0): undefined reference to `esp_image_basic_verify'
.pioenvs/esp32dev/libapp_update.a(esp_ota_ops.o): In function `esp_ota_end':
/Users/djmanas/.platformio/packages/framework-espidf/components/app_update/esp_ota_ops.c:233: undefined reference to `esp_image_basic_verify'
.pioenvs/esp32dev/libapp_update.a(esp_ota_ops.o): In function `esp_ota_set_boot_partition':
/Users/djmanas/.platformio/packages/framework-espidf/components/app_update/esp_ota_ops.c:374: undefined reference to `esp_image_basic_verify'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp32dev/firmware.elf] Error 1
```

**VSCode**: 1.22.2 (**MacOS High Sierra** 10.13.4)
**PlatformIO**:
> **Home**: 0.9.5
> **Core**: 3.5.3a9

**platformio.ini**
```
[platformio]
env_default=esp32dev

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf

[env:wemosbat]
platform = espressif32
board = wemosbat
framework = espidf
```

ivankravets, 2018-05-09:
Please reopen if you still need help.

Also, see http://docs.platformio.org/en/latest/platforms/espressif32.html#over-the-air-ota-update

closed 2018-05-09 by ivankravets

DJManas, 2018-05-09:
I am sorry for late reactions theese days. Work, family and stuff.

Had opened VSCode, opened the project.

Changed the **platform** to **espressif32** since I had seen, there had been an upgrade of the espressif32 platform in PIO.

Run:
```pio run --target clean```

Run:
```pio run```

Result:
**[SUCCESS] Took 197.47 seconds**

Thanks for support.