platformio/platform-espressif32 #83: flash_size and upload maximum_size in platformio.ini

vanniaz, 2018-05-08:
Hello,

I see that for esp32-based boards the "upload" section of the platformio.ini file declares a flash size of 4MB but a maximum size of around 1MB:
  "upload": {
    "flash_size": "4MB",
    "maximum_size": 1310720,

while for esp8266-based boards there is no "flash_size" entry but "maximum_size" is declared to be 4MB:
  "upload": {
    "maximum_size": 4194304,

The examples above are taken from esp32dev.json and nodemcuv2.json.
Note that 1310720 is not a power of 2, while 4194304 is 2^22.
Moreover, in the "board explorer" all esp32 based boards are listed with a ROM size of 1MB, despite the fact that many of these use 4MB flash chips.
Can you help me understand all this?
Thanks
Giovanni

BorisFR, 2018-05-09:
It is due to partition of the memory. Have a look at issue #58 .

vanniaz, 2018-05-10:
@BorisFR 
Ok thanks, now it is clear.
Perhaps in the board explorer it would be better to indicate the flash size, not the partition size (all ESP32 boards are listed as "1MB" flash, also in the online table: https://platformio.org/boards)

closed 2018-05-10 by vanniaz