platformio/platform-espressif32 #86: Compile error with latest development version

lucadentella, 2018-05-14:
Hi

new project, changed the platformio.ini file to use the development version:

    [env:lolin32]
    platform = https://github.com/platformio/platform-espressif32.git
    board = lolin32
    framework = espidf

the compile process fails with the following error:
```
In file included from C:/Users/luca/.platformio/packages/framework-espidf/components/mbedtls/port/include/mbedtls/esp_config.h:2665:0,
from C:/Users/luca/.platformio/packages/framework-espidf/components/mbedtls/include/mbedtls/sha256.h:29,
from C:\Users\luca\.platformio\packages\framework-espidf\components\bootloader_support\src\bootloader_sha.c:22:
C:/Users/luca/.platformio/packages/framework-espidf/components/mbedtls/include/mbedtls/check_config.h:73:2: error: #error "MBEDTLS_CTR_DRBG_C defined, but not all prerequisites"
#error "MBEDTLS_CTR_DRBG_C defined, but not all prerequisites"
^
*** [.pioenvs\lolin32\bootloader_support\src\bootloader_sha.o] Error 1
```

lucadentella, 2018-05-14:
I also noticed that, everytime I run the compile process, it says:

    Warning! Detected an outdated "sdkconfig.h" file. The old "sdkconfig.h" will be replaced by the new one.

and creates a .bak file... even if I delete it

ivankravets, 2018-05-14:
Please run `pio update` in PIO IDE for VSCode. It should work now.

closed 2018-05-14 by ivankravets

lucadentella, 2018-05-15:
fixed, thanks!

commit 2096d5b3 (2018-05-14): Use the actual dev/platform directory when looking for "sdkconfig.h" // Resolve #86
commit a45da6ac (2018-05-14): Merge branch 'develop' into feature/stage

* develop:
  Use the actual dev/platform directory when looking for "sdkconfig.h" // Resolve #86