platformio/platform-espressif32 #87: Several examples not compiling after fresh install of platformio

jcwren, 2018-05-23:
I attempted to compile the ```espidf-peripherals-uart``` example. I started with a fresh Linux box that did not have ```platformio``` installed, and installed it according to the instructions. I then followed the instructions in the example, and it's complaining about ```GPIO_NUM_4``` and ```GPIO_NUM_5``` not being declared. 

Also tried the ```espidf-ble-adv``` example. It's unhappy that ```esp_bt.h``` isn't found, and to run ```platformio lib search "header:esp_bt.h"```, which reports that nothing has been found. I've included the installation process and error text from the ```espidf-peripherals-uart``` example.

```
jcw@bark:~$ sudo easy_install pip
Searching for pip
Reading https://pypi.python.org/simple/pip/
Best match: pip 10.0.1
Downloading https://files.pythonhosted.org/packages/ae/e8/2340d46ecadb1692a1e455f13f75e596d4eab3d11a57446f08259dee8f02/pip-10.0.1.tar.gz#sha256=f2bd08e0cd1b06e10218feaf6fef299f473ba706582eb3bd9d52203fdbd7ee68
Processing pip-10.0.1.tar.gz
Writing /tmp/easy_install-e0iNTj/pip-10.0.1/setup.cfg
Running pip-10.0.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-e0iNTj/pip-10.0.1/egg-dist-tmp-NynOys
/usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
  warnings.warn(msg)
warning: no previously-included files found matching '.coveragerc'
warning: no previously-included files found matching '.mailmap'
warning: no previously-included files found matching '.travis.yml'
warning: no previously-included files found matching '.landscape.yml'
warning: no previously-included files found matching 'src/pip/_vendor/Makefile'
warning: no previously-included files found matching 'tox.ini'
warning: no previously-included files found matching '*-requirements.txt'
warning: no previously-included files found matching 'appveyor.yml'
warning: no previously-included files found matching 'src/pip/_vendor/six'
warning: no previously-included files matching '*.pyi' found under directory 'src/pip/_vendor'
no previously-included directories found matching '.github'
no previously-included directories found matching '.travis'
no previously-included directories found matching 'docs/build'
no previously-included directories found matching 'news'
no previously-included directories found matching 'contrib'
no previously-included directories found matching 'tasks'
no previously-included directories found matching 'tests'
creating /usr/local/lib/python2.7/dist-packages/pip-10.0.1-py2.7.egg
Extracting pip-10.0.1-py2.7.egg to /usr/local/lib/python2.7/dist-packages
Adding pip 10.0.1 to easy-install.pth file
Installing pip script to /usr/local/bin
Installing pip2.7 script to /usr/local/bin
Installing pip2 script to /usr/local/bin

Installed /usr/local/lib/python2.7/dist-packages/pip-10.0.1-py2.7.egg
Processing dependencies for pip
Finished processing dependencies for pip
jcw@bark:~$ sudo pip install -U platformio
The directory '/home/jcw/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/jcw/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting platformio
  Downloading https://files.pythonhosted.org/packages/d2/4a/0a69838890fdabe4603fdb3c1bc6c77e8f20e8b7391a21643505214a907c/platformio-3.5.2-py27-none-any.whl (142kB)
    100% |████████████████████████████████| 143kB 5.1MB/s
Collecting lockfile<0.13,>=0.9.1 (from platformio)
  Downloading https://files.pythonhosted.org/packages/c8/22/9460e311f340cb62d26a38c419b1381b8593b0bb6b5d1f056938b086d362/lockfile-0.12.2-py2.py3-none-any.whl
Collecting colorama (from platformio)
  Downloading https://files.pythonhosted.org/packages/db/c8/7dcf9dbcb22429512708fe3a547f8b6101c0d02137acbd892505aee57adf/colorama-0.3.9-py2.py3-none-any.whl
Collecting semantic-version<3,>=2.5.0 (from platformio)
  Downloading https://files.pythonhosted.org/packages/72/83/f76958017f3094b072d8e3a72d25c3ed65f754cc607fdb6a7b33d84ab1d5/semantic_version-2.6.0.tar.gz
Collecting bottle<0.13 (from platformio)
  Downloading https://files.pythonhosted.org/packages/bd/99/04dc59ced52a8261ee0f965a8968717a255ea84a36013e527944dbf3468c/bottle-0.12.13.tar.gz (70kB)
    100% |████████████████████████████████| 71kB 8.6MB/s
Collecting requests<3,>=2.4.0 (from platformio)
  Downloading https://files.pythonhosted.org/packages/49/df/50aa1999ab9bde74656c2919d9c0c085fd2b3775fd3eca826012bef76d8c/requests-2.18.4-py2.py3-none-any.whl (88kB)
    100% |████████████████████████████████| 92kB 8.6MB/s
Collecting click<6,>=5 (from platformio)
  Downloading https://files.pythonhosted.org/packages/8f/98/14966b6d772fd5fba1eb3bb34a62a7f736d609572493397cdc5715c14514/click-5.1-py2.py3-none-any.whl (65kB)
    100% |████████████████████████████████| 71kB 11.2MB/s
Requirement not upgraded as not directly required: pyserial!=3.3,<4,>=3 in /usr/lib/python2.7/dist-packages (from platformio) (3.0.1)
Collecting certifi>=2017.4.17 (from requests<3,>=2.4.0->platformio)
  Downloading https://files.pythonhosted.org/packages/7c/e6/92ad559b7192d846975fc916b65f667c7b8c3a32bea7372340bfe9a15fa5/certifi-2018.4.16-py2.py3-none-any.whl (150kB)
    100% |████████████████████████████████| 153kB 8.0MB/s
Collecting chardet<3.1.0,>=3.0.2 (from requests<3,>=2.4.0->platformio)
  Downloading https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl (133kB)
    100% |████████████████████████████████| 143kB 8.2MB/s
[... 156 lines cut ...]
src/main.c:26:25: note: each undeclared identifier is reported only once for each function it appears in
#define ECHO_TEST_TXD  (GPIO_NUM_4)
^
src/main.c:45:30: note: in expansion of macro 'ECHO_TEST_TXD'
uart_set_pin(UART_NUM_1, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS);
^
src/main.c:27:25: error: 'GPIO_NUM_5' undeclared (first use in this function)
#define ECHO_TEST_RXD  (GPIO_NUM_5)
^
src/main.c:45:45: note: in expansion of macro 'ECHO_TEST_RXD'
uart_set_pin(UART_NUM_1, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS);
^
Compiling .pioenvs/esp32dev/bootloader/bootloader_start.o
Compiling .pioenvs/esp32dev/bootloader/flash_qio_mode.o
Compiling .pioenvs/esp32dev/bootloaderSupport/src/bootloader_flash.o
*** [.pioenvs/esp32dev/src/main.o] Error 1
================================================================================================================== [ERROR] Took 1.63 seconds ==================================================================================================================

========================================================================================================================== [SUMMARY] ==========================================================================================================================
Environment nano32      [ERROR]
Environment espea32     [ERROR]
Environment esp32dev    [ERROR]
================================================================================================================== [ERROR] Took 23.19 seconds ==================================================================================================================
jcw@bark:~/platform-espressif32-develop/examples/espidf-peripherals-uart$
```

jcwren, 2018-05-23:
Added ```#include <driver/gpio.h>``` , gets past that error, blows up with

```
jcw@jcw-PC ~/Downloads/platform-espressif32-develop/examples/espidf-peripherals-uart
$ platformio run
[05/23/18 17:14:09] Processing nano32 (platform: espressif32; board: nano32; framework: espidf)

Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > MakerAsia Nano32
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 0 compatible libraries
Scanning dependencies...
No dependencies
Archiving .pioenvs\nano32\libbootloaderMicroEcc.a
Compiling .pioenvs\nano32\bootloaderSoc\esp32\brownout.o
Indexing .pioenvs\nano32\libbootloaderMicroEcc.a
Compiling .pioenvs\nano32\bootloaderSoc\esp32\rtc_clk.o
Compiling .pioenvs\nano32\bootloaderSoc\esp32\rtc_init.o
Compiling .pioenvs\nano32\bootloaderSoc\esp32\rtc_pm.o
Compiling .pioenvs\nano32\bootloaderSoc\esp32\rtc_sleep.o
Compiling .pioenvs\nano32\bootloaderSoc\esp32\rtc_time.o
In file included from C:\cygwin\home\jcw\.platformio\packages\framework-espidf\components\soc\esp32\brownout.c:21:0:
C:\cygwin\home\jcw\.platformio\packages\framework-espidf\components\soc\esp32\brownout.c: In function 'esp_brownout_init':
C:\cygwin\home\jcw\.platformio\packages\framework-espidf\components\soc\esp32\brownout.c:34:45: error: 'CONFIG_BROWNOUT_DET_RESETDELAY' undeclared (first use in this function)
RTC_CNTL_BROWN_OUT_RST_ENA | (((CONFIG_BROWNOUT_DET_RESETDELAY*150)/1000) << RTC_CNTL_BROWN_OUT_RST_WAIT_S) |
^
C:/cygwin/home/jcw/.platformio/packages/framework-espidf/components/soc/esp32/include/soc/soc.h:218:76: note: in definition of macro 'WRITE_PERI_REG'
(*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val);                                       \
^
C:\cygwin\home\jcw\.platformio\packages\framework-espidf\components\soc\esp32\brownout.c:34:45: note: each undeclared identifier is reported only once for each function it appears in
RTC_CNTL_BROWN_OUT_RST_ENA | (((CONFIG_BROWNOUT_DET_RESETDELAY*150)/1000) << RTC_CNTL_BROWN_OUT_RST_WAIT_S) |
^
C:/cygwin/home/jcw/.platformio/packages/framework-espidf/components/soc/esp32/include/soc/soc.h:218:76: note: in definition of macro 'WRITE_PERI_REG'
(*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val);                                       \
^
*** [.pioenvs\nano32\bootloaderSoc\esp32\brownout.o] Error 1
 [ERROR] Took 4.55 seconds
jcw@jcw-PC ~/Downloads/platform-espressif32-develop/examples/espidf-peripherals-uart
$
```

ivankravets, 2018-05-25:
It seems that you use example from dev/branch but the source code of dev/platform is from stable branch. Could you try the same example with the latest/upstream version?

http://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions

jcwren, 2018-05-29:
I won't be able to try this for another couple of days, but I will add that I didn't intentionally choose one branch or the other, just followed the instructions.

ivankravets, 2018-05-31:
The final release is out! Please update your dev/platform.

closed 2018-05-31 by ivankravets