platformio/platform-espressif32 #89: Error uploading: 'PlatformBoardConfig' object has no attribute 'update'

xoseperez, 2018-05-31:
I just updated the platform to the latest stable one and I'm getting this error when uploading the sketch to the board:

```
$ pio run -t upload
[Thu May 31 18:57:17 2018] Processing lolin32 (platform: espressif32; board: lolin32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > WEMOS LOLIN32
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
DEBUG: CURRENT(minimodule) EXTERNAL(minimodule, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 21 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Ticker> v1.1
|-- <WiFi> v1.0
|-- <ArduinoOTA> v1.0
|   |-- <WiFi> v1.0
|   |-- <Update> v1.0
|   |-- <ESPmDNS> v1.0
|   |   |-- <WiFi> v1.0
AttributeError: 'PlatformBoardConfig' object has no attribute 'update':
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/main.py", line 162:
env.SConscript("$BUILD_SCRIPT")
File "/home/xose/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/xose/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/home/xose/.platformio/platforms/espressif32/builder/main.py", line 249:
_update_max_upload_size(env)
File "/home/xose/.platformio/platforms/espressif32/builder/main.py", line 89:
env.BoardConfig().update("upload.maximum_size", max(sizes))
```
Status:

```
$ pio --version
PlatformIO, version 3.5.2
$ pio update
Updating tool-scons                      @ 2.20501.4      [Up-to-date]
Updating contrib-pysite                  @ 0.2.0          [Up-to-date]
Updating tool-pioplus                    @ 1.3.1          [Up-to-date]

Platform Manager
================
(...)
Platform Espressif 32
--------
Updating espressif32                     @ 1.0.0          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20310.0      [Up-to-date]
Updating tool-espotapy                   @ 1.1.0          [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ 1.5.3          [Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]
(...)
```

vijaygill, 2018-05-31:
Same issue with "pio run -e esp32dev -t upload" here.

ivankravets, 2018-05-31:
Oh... We depend on the PIO Core 3.5.3 which has not been released yet :( `pio upgrade --dev` should fix this issue but that is not a solution.

I'll take a look what do we need to have a final release of PIO Core.

ivankravets, 2018-06-01:
PIO Core 3.5.3 is out! Please `pio upgrade`

- https://github.com/platformio/platformio-core/releases/tag/v3.5.3

closed 2018-06-01 by ivankravets

xoseperez, 2018-06-02:
Upgrading fixes the issue. Thank you!