platformio/platform-espressif32 #90: PPP components missing from lwip in esp-idf

dmitrygaa, 2018-06-04:
My project, using PPP portion of lwip stack, failed to build with PlatformIO, but it can be successfully built via ESP32 toolchain and esp-idf SDK 3.0.
Build fails with error:
```
.pioenvs\esp32dev\liblwip.a(init.o):(.literal.lwip_init+0x0): undefined reference to `ppp_init'
.pioenvs\esp32dev\liblwip.a(init.o): In function `lwip_init':
init.c:(.text.lwip_init+0x15): undefined reference to `ppp_init'
```
Some examination shows that folder componets/lwip/netif/ppp is not included in the build.

Environment:
VS Code
PlatformIO 3.6.0a1
ESP32 platform 1.0.2

the-blest, 2019-04-10:
Seems like a problem in [/builder/frameworks/espidf.py](https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/espidf.py). It will parse some of the makefiles including the one for lwip to find out what needs to be built.
In my local install it's here: _.platformio/packages/framework-espidf/components/lwip/component.mk_. 
However this file has a variable that is sometimes assigned dependent on the compile time define **CONFIG_PPP_SUPPORT**. The variable is not evaluated in _espidf.py_. This is the problematic section of the makefile where **LWIP_PPP_DIRS** is sometimes assigned, but the script does not add it to **COMPONENT_SRCDIRS**.
```
ifdef CONFIG_PPP_SUPPORT
LWIP_PPP_DIRS := netif/ppp/polarssl netif/ppp
else
LWIP_PPP_DIRS :=
endif

COMPONENT_SRCDIRS := \
	api \
	apps apps/sntp apps/ping \
	core core/ipv4 core/ipv6 \
	$(LWIP_PPP_DIRS) netif \
	port/freertos port/netif port/debug port 
```
The end result is that regardless of the define the PPP files are not compiled. My temporary local workaround is to remove the conditional assignment and forcefully enable PPP all the time, since I only have one active project atm this works for me. I don't have a good suggestion for a permanent fix yet though.

the-blest, 2019-07-01:
CONFIG_PPP_SUPPORT is now taken into account in the python builder as well. It was added in [this commit](https://github.com/platformio/platform-espressif32/commit/26605923f1de1895dfefd50f71fc6d88f718ae4b) and took effect in release [v1.8.0](https://github.com/platformio/platform-espressif32/releases/tag/v1.8.0) when ESP-IDF was updated to 3.2.0.

valeros, 2019-09-18:
Feel free to reopen the issue if you still need help.

closed 2019-09-18 by valeros