platformio/platform-espressif32 #91: Invalid paths to cJSON library component of esp-idf

dmitrygaa, 2018-06-04:
While building rather simple project, using cJSON, I've got error message:
```
components/utils/JSON.h:4:19: fatal error: cJSON.h: No such file or directory
```
This projects successfully builds with ESP32 toolchain and esp-idf SDK 3.0.
When I look in .pioenvs/esp32dev/longcmd-* files I see these folders in include search dirs:
```
.platformio/packages/framework-espidf/components/json/include
.platformio/packages/framework-espidf/components/json/port/include
```
But there are no /include or /port/include dirs inside json component. The only dir is cJSON. You can check directory structure [here](https://github.com/espressif/esp-idf/tree/release/v3.0/components/json).
To temporarily solve this issue, I've had to change include to:
```
#include "../../json/cJSON/cJSON.h"
```
Environment:
VSCode
PlatformIO 3.6.0a1
esp-idf platform 1.0.2

wegunterjr, 2018-08-25:
I had the same problem. I followed your suggestion. So far so good. Thanks!

ivankravets, 2018-10-01:
Resolved in https://github.com/platformio/platform-espressif32/commit/bcfc6a4913463c5ab434e742b7d3208e393ede51

closed 2018-10-01 by ivankravets