platformio/platform-espressif32 #93: m5stack program size is greater than maximum allowed

zsoltmazlo, 2018-06-07:
I have a project which I have developed in my previous configuration (ubuntu 14.04, visual studio code), and I wanted to continue it in my new config (ubuntu 18.04, clion). To prevent any problem that can come up by changing the IDE, I created a new project, initialized with pio command to have a clion-compatible project, and then I have copied all sources to the new project from the old one. Everything went fine until I wanted to upload the code onto my m5stack watch when I got the following error:

> Error: The program size (1338029 bytes) is greater than maximum allowed (1310720 bytes)

As I figured it out from the [M5 core schematic](http://www.m5stack.com/download/M5-Core-Schematic(20171206).pdf), M5 has an [GD25Q32C](http://www.gigadevice.com/wp-content/uploads/2017/12/DS-00088-GD25Q32C-Rev3.5.pdf) Serial flash, which is 32M-bit big - thus, I tried to configure the flash size in different ways:

- Added `board_upload.maximum_size = 4194304` to the .ini file which was recognized by the compiler:
> PLATFORM: Espressif 32 > M5Stack Core ESP32
> SYSTEM: ESP32 240MHz 288KB RAM (4MB Flash)
- Searched board-config file in platformio folder (~/.platformio/platforms/espressif32/boards/m5stack-core-esp32.json) and changed the  `"maximum_size: 1310720"` line to  `"maximum_size": 4194304`
- Searched boards.txt file in platformio folder (~/.platformio/packages/framework-arduinoespressif32/boards.txt) and changed the `m5stack-core-esp32.upload.maximum_size=1310720` line to `m5stack-core-esp32.upload.maximum_size=4194304`.

Not even these miserable steps fixed my issue. 

Then I tried to upload my previous project with pio command from command-line, but I had to face with the very same error.

What am I doing wrong and how can I fix this? I think the config json and the boards.txt should not be edited manually, but at least I have tried.

zsoltmazlo, 2018-06-07:
Thank you for your quick answer, it did solved my problem!

closed 2018-06-07 by zsoltmazlo

ivankravets, 2018-06-07:
How did you solve it?

zsoltmazlo, 2018-06-07:
The problem was that if you don't explicitly name the partition csv, then it use a default one (I'm not sure which one), and in that csv app space was defined to be no greater than 1.25MB. I made a custom partition.csv and now I have almost 3.5MB app space (I'm sort of started to get familiar with spiffs).

hanadjerbi1, 2018-08-03:
where i can find the file .csv ?

hanadjerbi1, 2018-08-03:
did you creat the file partition.csv or you make a modification ?

ayoubgoun, 2020-02-28:
1. go to PlatformIO.ini 
2. add         board_build.partitions = no_ota.csv

thefatmoop, 2020-11-02:
Just some additional help for others in this predicament: If you do need OTA, but you've just run out of space - SPIFFS has a monstrously large partition size. I wasn't using very much spiffs, so i shrunk that partition down and moved the free space to the other two partitions. Why is so much space allocated to SPIFFS by default? 

Anyway, here's my partition layout, there's still some spiffs. 
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1C0000,
app1,     app,  ota_1,   0x1D0000,0x1C0000,
spiffs,   data, spiffs,  0x390000,0x70000,

Just a heads up - you can't OTA update the partition table. You'll need to re-flash every device over serial, then OTA will work again.

Ibrahim2595, 2021-11-08:
Hi - 

I am trying to compile an ESP32 Cam file via platformio and have an external library the I exported from Edge Impulse.
After Arduino kept giving me errors with "File name is too long" I decided to move to platformio. So here I am.
I managed to fix all bugs along the process, but now stuck with this:

Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (2536764 bytes) is greater than maximum allowed (2097152 bytes)

Reading the forum above, I did the below:
go to PlatformIO.ini
add board_build.partitions = no_ota.csv

[image]
[image]
[image]

valeros, 2021-11-09:
@Ibrahim2595 It seems you need to further increase the partition size for you app, try this one:
```ini
board_build.partitions = huge_app.csv
```