platformio/platform-espressif32 #94: Feature Request: Import existing ESP32 IDF based project

joaofl, 2018-06-11:
Right now, the platform IO only allows to create a new project based on the ESP32 IDF, but not import an existing one. That would be of great value.

Asalways2, 2019-04-18:
Is there a manual way of importing an ESP-IDF project with Platformio?
I cloned my ESP-IDF PIO project to another PC but can't get it to build.

JayNaire, 2019-04-23:
## My scrappy notes on getting esp-idf examples running with pio 23Apr19

### Locations of esp-idf
~/.platformio/packages/framework-espidf/
~/esp/esp-idf (orignal esp-idf installation)
Which location is used for **make menuconfig** depends on **$IDF_PATH**

### Getting ESP-IDF examples working with platformio

  * Copy an example from **~/.platformio/packages/framework-espidf/examples**
  eg: ~/tmp/mcpwm/mcpwm_basic_config

  ~~In a terminal **cd to ~/tmp/mcpwm/mcpwm_basic_config** set \$IDF_PATH~~
  NOT REQUIRED
  ```
  cd /tmp/mcpwm/mcpwm_basic_config
  IDF_PATH=~/.platformio/packages/framework-espidf/
  export IDF_PATH
  ```
  *  ~~run **make menuconfig**~~ (works but can't be used by platformio - gotta stick with pio's default sdkconfig.h :( see below)
  * Open the folder in vscode
  * create platformio.ini in top level folder from this text (modify it first for your environment!):

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = espidf

monitor_speed = 115200

;serial
upload_speed = 115200
upload_port = /dev/ttyUSB0

;OTA
; upload_protocol = espota
; upload_port = GasMonitor.local

lib_deps =
     FreeRTOS@>=10.2.0-2

; src filters
; see http://docs.platformio.org/en/stable/projectconf/section_env_build.html#src-filter

```
* Now ~~***either***~~ remove mcpwm_basic_config from the workspace and re-open it to get platformio to read the new platformio.ini. ~~***Or***, from terminal, **platformio init --board esp32doit-devkit-v1 --platform espressif32 --framework espidf**~~
* Check platformio has made .pioenvs etc
* Make sure there is a src folder (examples use main so just rename or link it **ln -s main/ src**)
* Build
* A default, unsorted, sdkconfig.h will automatically be added to src folder at build time but it has no relationship to sdkconfig that **make menuconfig** produces :( 
* Remove all comments from sdkconfig.h
* Run sdkconfig.h through sort to make things easier to find. **DO NOT ADD ANY COMMENTS** to sdkconfig.h. Comments can foul up the generation of .pioenvs/\<board>/esp32_out.ld and linking fails.

jack001214, 2019-07-25:
@JayNaire  what about libraries?

valeros, 2020-05-25:
Closing as the latest espressif32 platform uses native CMake build system https://docs.platformio.org/en/latest/frameworks/espidf.html#project-structure

closed 2020-05-25 by valeros